MarlinFirmware/Marlin #24728: [BUG] Servo probe does not deploy during LCD bed tramming despite BED_TRAMMING_USE_PROBE being enabled

Loki44, 2022-09-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I activate bed tramming using the LCD menu, the printer homes correctly, deploying the probe for homing Z, but then goes to the first corner and just starts lowering the gantry, without deploying the probe.  I have "BED_TRAMMING_USE_PROBE" enabled in configuration.h.

### Expected behavior

the probe to be deployed for tramming.

### Actual behavior

the printer just slowly lowers the gantry without deploying the probe.

### Steps to Reproduce

1. configure servo probe on printer
2. enable LCD_BED_TRAMMING in configuration.h
3. enable BED_TRAMMING_USE_PROBE in configuration.h
4. activate bed tramming

### Version of Marlin Firmware

2.1.1 and bugfix-2.1.x

### Printer model

heavily modified tevo tarantula

### Electronics

mks gen 1.4

### Add-ons

servo probe

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9484549/Configuration.zip)

Loki44, 2022-11-14:
> This issue has had no activity in the last 60 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 10 days.

a reply

atharvbadkas, 2023-01-20:
I also face this issue on my both printers.....I have to deploy the arm(rotate the servo arm) manually for LCD Bed Level Corners

atharvbadkas, 2023-01-20:
Also LCD Bed Tramming seems broken in latest 2.1.2 version.
Probe moves out of the bed.
I tried same probe and tramming offset settings in my last bugfix 2.0.x build, it works.
Heres link for config files both 2.1.2 and bugfix 2.0.x
https://drive.google.com/drive/folders/1LnuIhi-lhzPGr8iu1hqYCGzM94UyJwDP?usp=sharing

Farva42, 2023-02-18:
I've taken a look at trying to get it to do a probe deploy at the start and a probe stow at the end but as soon as the probe deploy is added it will start looping after the final point is probed it moves to stow the probe and then goes back to the first point and starts over. I've tried to pin down what is causing the loop but so far I haven't been able to. I turned on bed leveling debugging and it's not showing a probe deploy when it starts looping. I added a serial echo at the start of the probing function and it's not echoing till it probes the second point after it starts looping. I've also tried stripping out the homing checks and the ending select screen to go straight to the status screen and it will still loop. 

It looks like it is ignoring probe x&y offsets as well. Which is why it's probing off the bed. 

There's also an issue using a DWIN screen where the last z hieght is not updating correctly as well as the current number of good points. The screen does not clear the old numbers and just writes over the old numbers.

thisiskeithb, 2023-02-24:
A PR has been opened, so closing:

- https://github.com/MarlinFirmware/Marlin/pull/25425

closed 2023-02-24 by thisiskeithb