MarlinFirmware/Marlin #24729: [FR] Insert SV04 Creality Board

ils15, 2022-09-04:
### Are you looking for hardware support?

Hi.
Sovol SV04 IDEX uses a custom board based on CrealityV4 in your IDEX printer
I tried several times to compile but original CrealityV4 didnt accept more than 1 extruder in Marlin
I'm Searching the best way to add this board in Marlin (i made some pins edits here) but i'm a little bit confused how to make it

Here is the source code for Board and Marlin 
https://drive.google.com/file/d/1mkSEq5V3vXLtwY_EB5MizB4gfCME56gj/view

Board name

Creality V5.2.1

### Describe the feature you want

New Board Creality for SV04 IDEX

### Additional context

[image]
Board screenshot

hcet14, 2022-09-06:
['https://www.serial-testeur.com/en/sovol-sv04-test-of-the-new-idex-dual-extruders-printer/'](https://www.serial-testeur.com/en/sovol-sv04-test-of-the-new-idex-dual-extruders-printer/)
or ask Sovol to sponsor Marlin. They make money with that printer. Nothing in life is for free, maybe death ;). You may ask Creality to support your Creality V5.2.1?

ellensp, 2022-09-14:
Since a PR to add this has been created, closing this request

closed 2022-09-14 by ellensp