MarlinFirmware/Marlin #24730: [FR] Host access to SD card on SKR 3

FlyingSamson, 2022-09-04:
### Describe the feature you want

It would be great to be able to access the SD card from a host connected via USB to the SKR 3 board similar to #22354.

It seems SDIO support was made available in #24271, but there is no `*_USB` environment in `stm32h7.ini` for the SKR 3 and using `env:STM32H743Vx_btt` with `#define SDSUPPORT` enabled in `Configuration.h` and `#define SDCARD_CONNECTION ONBOARD` enabled in `Configuration_adv.h` I do not see the SD card when connecting the board to a computer (neither on Mac OSX nor Raspbian).

I tried adapting the environment similar to the changes in #22354 but was unable to get it to compile, probably because I don't really know what I'm doing. I suspect one problem might be, that `${stm_flash_drive.platform_packages}` requires an older version of `ststm32` than is requested in `stm32h7.ini` for the SKR 3 board.

thisiskeithb, 2022-09-04:
> I suspect one problem might be, that ${stm_flash_drive.platform_packages} requires an older version of ststm32 than is requested in stm32h7.ini for the SKR 3 board.

That and the SKR 3 does not have a USB A socket for flash drive support.

sl1pkn07, 2022-09-07:
I'm not sure if is a bug/related. but i also can't show the SD over SD in my ReARM on my linux (i don't have mac/win for test)

the unit is recognized as "Marlin SDcard 01" device. 

[image]

but i get a tons of errors in my linux tty

~~~dmesg
[11618.592405] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[11618.733779] usb 1-3: New USB device found, idVendor=1d50, idProduct=6029, bcdDevice= 1.00
[11618.733792] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11618.733796] usb 1-3: Product: Marlin USB Device
[11618.733800] usb 1-3: Manufacturer: marlinfw.org 
[11618.733803] usb 1-3: SerialNumber: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
[11618.780918] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[11618.780948] usbcore: registered new interface driver cdc_acm
[11618.780950] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[11618.782540] usb-storage 1-3:1.2: USB Mass Storage device detected
[11618.782717] scsi host13: usb-storage 1-3:1.2
[11618.782859] usbcore: registered new interface driver usb-storage
[11618.788496] usbcore: registered new interface driver uas
[11619.789483] scsi 13:0:0:0: Direct-Access     Marlin   SDCard 01        1.0  PQ: 0 ANSI: 0 CCS
[11619.789755] sd 13:0:0:0: Attached scsi generic sg6 type 0
[11619.790087] sd 13:0:0:0: [sdf] 245760 512-byte logical blocks: (126 MB/120 MiB)
[11619.790383] sd 13:0:0:0: [sdf] Write Protect is off
[11619.790386] sd 13:0:0:0: [sdf] Mode Sense: 00 00 00 00
[11619.790676] sd 13:0:0:0: [sdf] Asking for cache data failed
[11619.790681] sd 13:0:0:0: [sdf] Assuming drive cache: write through
[11622.626811] sd 13:0:0:0: [sdf] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=2s
[11622.626819] sd 13:0:0:0: [sdf] tag#0 Sense Key : Not Ready [current] 
[11622.626822] sd 13:0:0:0: [sdf] tag#0 Add. Sense: Medium not present
[11622.626825] sd 13:0:0:0: [sdf] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11622.626835] I/O error, dev sdf, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[11622.626844] Buffer I/O error on dev sdf, logical block 0, async page read
[11622.627334] sd 13:0:0:0: [sdf] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=0s
[11622.627337] sd 13:0:0:0: [sdf] tag#0 Sense Key : Not Ready [current] 
[11622.627339] sd 13:0:0:0: [sdf] tag#0 Add. Sense: Medium not present - loadable
[11622.627341] sd 13:0:0:0: [sdf] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11622.627344] I/O error, dev sdf, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[11622.627348] Buffer I/O error on dev sdf, logical block 0, async page read
[11622.627354] ldm_validate_partition_table(): Disk read failed.
[11622.627774] sd 13:0:0:0: [sdf] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=0s
[11622.627776] sd 13:0:0:0: [sdf] tag#0 Sense Key : Not Ready [current] 
[11622.627778] sd 13:0:0:0: [sdf] tag#0 Add. Sense: Medium not present - loadable
[11622.627780] sd 13:0:0:0: [sdf] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11622.627781] I/O error, dev sdf, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[11622.627784] Buffer I/O error on dev sdf, logical block 0, async page read
[11622.628210] sd 13:0:0:0: [sdf] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=0s
[11622.628212] sd 13:0:0:0: [sdf] tag#0 Sense Key : Not Ready [current] 
[11622.628214] sd 13:0:0:0: [sdf] tag#0 Add. Sense: Medium not present - loadable
[11622.628216] sd 13:0:0:0: [sdf] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11622.628218] I/O error, dev sdf, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[11622.628220] Buffer I/O error on dev sdf, logical block 0, async page read
[11622.628660] sd 13:0:0:0: [sdf] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=0s
[11622.628662] sd 13:0:0:0: [sdf] tag#0 Sense Key : Not Ready [current] 
[11622.628664] sd 13:0:0:0: [sdf] tag#0 Add. Sense: Medium not present - loadable
[11622.628666] sd 13:0:0:0: [sdf] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11622.628667] I/O error, dev sdf, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[11622.628669] Buffer I/O error on dev sdf, logical block 0, async page read
[11622.628675]  sdf: unable to read partition table
[11622.628973] sd 13:0:0:0: [sdf] Attached SCSI removable disk
[11622.630105] sdf: detected capacity change from 245760 to 0
~~~

but the SD works ok throught USB SD card reader

'm not sure if is a filesystem (Fat32 if i remember) problem

the option https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1711 is enabled

greetings

razserv2010, 2022-10-03:
I agree with the request to add this option to the SKR3 EZ board please..

carlosjln, 2022-12-17:
Follow up on this, a much needed feature. 👏

These should be standard features that must be implemented across all boards/firmware versions.

TheRavenV6, 2023-01-01:
I'd like to add a request for USB mass storage access through USB for the SKR3 as well.

thinkyhead, 2023-01-17:
If the following `build_flags` are added to the `STM32H743Vx_btt` environment in `stm32h7.ini` does it help?
```
-DUSE_USB_HS_IN_FS -DUSE_USBHOST_HS -DUSBD_USE_CDC_MSC
-DUSBD_IRQ_PRIO=5 -DUSBD_IRQ_SUBPRIO=6
```

thinkyhead, 2023-01-17:
To get this functionality, we'll need an upgrade to https://github.com/rhapsodyv/Arduino_Core_STM32/tree/usb-host-msc-cdc-msc-3 … with a little guidance from @rhapsodyv. I looked over the differences between the previous Arduino_Core_STM32 and the new one, and there are some refactors in there, so it looks like some changes might be needed to get it all functioning.

nfored, 2023-01-27:
I also would like this since we can't use usb to push firmware. Would love to not have to open the printer every time.

nowkillkennys, 2023-03-25:
same same here skr 3 ez marlin 2.1.x bugfix

Fabface, 2023-12-16:
is this supposed to be working now? I can't access the SD over USB (Marlin 2.1.2.1)

thisiskeithb, 2023-12-17:
> is this supposed to be working now?

No, this feature request is still open.

fjulian79, 2024-08-28:
I want to push this request .. For me it's about firmware updates via octoprint

iStruggle, 2025-07-24:
I would also appreciate this feature - in my configuration, it was necessary to block access to the SD, so flashing requires dismantling.