MarlinFirmware/Marlin #24733: [BUG] Compilation error when activating AUTO_BED_LEVELING_UBL

dmidge8, 2022-09-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When activating the AUTO_BED_LEVELING_UBL as a leveling method, there is a compilation error that doesn't seem configuration related. I have no issue when I choose the bilinear bed leveling.

The eeprom is activated, as per specified in the documentation.

I am using Archlinux as an OS, Arduino IDE 1.8.19.

The stack of compile error is as follows:

```
In file included from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/inc/MarlinConfig.h:47,
                 from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl.cpp:23:
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/core/types.h: In function 'unified_bed_leveling::display_map(unsigned char)':
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/core/types.h:385:16: error: inlining failed in call to 'always_inline' 'XYval<float>::operator+(XYval<float> const&)': target specific option mismatch
  385 |   FI XYval<T>  operator+ (const XYval<T>   &rs)         { XYval<T> ls = *this; ls.x += rs.x; ls.y += rs.y; return ls; }
      |                ^~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl.cpp:189:77: note: called from here
  189 |   const xy_int8_t curr = closest_indexes(xy_pos_t(current_position) + probe.offset_xy);
      |                                                                             ^~~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/core/types.h:681:6: error: inlining failed in call to 'always_inline' 'XYZEval<float>::operator XYval<float>&()': target specific option mismatch
  681 |   FI operator       XYval<T>&()        { return *(XYval<T>*)this; }
      |      ^~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl.cpp:189:67: note: called from here
  189 |   const xy_int8_t curr = closest_indexes(xy_pos_t(current_position) + probe.offset_xy);
      |                                                                   ^
In file included from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/gcode.h:331,
                 from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:37:
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/parser.h: In function 'unified_bed_leveling::G29_parse_parameters()':
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/parser.h:249:28: error: inlining failed in call to 'always_inline' 'GCodeParser::has_value()': target specific option mismatch
  249 |   FORCE_INLINE static bool has_value() { return !!value_ptr; }
      |                            ^~~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1123:42: note: called from here
 1123 |     param.R_repetition = parser.has_value() ? parser.value_byte() : GRID_MAX_POINTS;
      |                          ~~~~~~~~~~~~~~~~^~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/parser.h:188:30: error: inlining failed in call to 'always_inline' 'GCodeParser::seen_test(char)': target specific option mismatch
  188 |     FORCE_INLINE static bool seen_test(const char c) { return TEST32(codebits, LETTER_BIT(c)); }
      |                              ^~~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1193:23: note: called from here
 1193 |   if (parser.seen_test('A')) {
      |       ~~~~~~~~~~~~~~~~^~~~~
In file included from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/inc/MarlinConfig.h:47,
                 from /home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:23:
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/core/types.h:331:13: error: inlining failed in call to 'always_inline' 'XYval<float>::set(float, float)': target specific option mismatch
  331 |     FI void set(const T px, const T py)                 { x = px; y = py; }
      |             ^~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1185:19: note: called from here
 1185 |   param.XY_pos.set(sx, sy);
      |   ~~~~~~~~~~~~~~~~^~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/parser.h:188:30: error: inlining failed in call to 'always_inline' 'GCodeParser::seen_test(char)': target specific option mismatch
  188 |     FORCE_INLINE static bool seen_test(const char c) { return TEST32(codebits, LETTER_BIT(c)); }
      |                              ^~~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1194:25: note: called from here
 1194 |     if (parser.seen_test('D')) {
      |         ~~~~~~~~~~~~~~~~^~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/gcode/parser.h:188:30: error: inlining failed in call to 'always_inline' 'GCodeParser::seen_test(char)': target specific option mismatch
  188 |     FORCE_INLINE static bool seen_test(const char c) { return TEST32(codebits, LETTER_BIT(c)); }
      |                              ^~~~~~~~~
/home/me/Workspace/Arduino/Prusa i3/github/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1201:28: note: called from here
 1201 |   else if (parser.seen_test('D')) {
      |            ~~~~~~~~~~~~~~~~^~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560. 
```

Attached, my configuration files.
[UblError.zip](https://github.com/MarlinFirmware/Marlin/files/9485468/UblError.zip)

### Version of Marlin Firmware

bugfix-2.1.x - 328f6d9affd01bbb33b865f444747a95c67a1ea2

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2022-09-04:
first - don't use arduino, use platformio
then see if error persist

Bob-the-Kuhn, 2022-09-05:
I just compiled via Auto Build Marlin in VSCode with no issues using the supplied configuration files.

ellensp, 2022-09-05:
bugfix 2.1.x + Ubuntu 22.04.1 LTS + Arduino IDE 1.8.19 + your configs.  Compiles without issues.

EvilGremlin, 2022-09-05:
So... when we gonna finally drop arduino?

closed 2022-10-14 by thisiskeithb