MarlinFirmware/Marlin #24735: After installation of Marlin, from Pronterface I receive "Unknown command"

thingAS, 2022-09-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi! This is very strange issue! I looked a lot in the internet for the same situation. 

I upload the Marin2.1.1 with platformio/VScode on Ramps1.4 with "SUCCESS"

I have two situations that I don´t know how/if they are related.

1 - I have a warning, from VSCode: #warning "Your Configuration provides no method to acquire user feedback!" 

2 - I can use Pronterface to move the printer but when I send a normal comand, for example, M20, it says Unknown command: "M20".

I tried 2 boards, other configurations. I can´t believe that only happen to me. Why? :) Please help!

### Bug Timeline

When I try install the new versions of Marlin with VSCode/Platformio

### Expected behavior

After instalations send comands and receive a feedback from the board (Ramps 1.4) of the printer.

### Actual behavior

The feedback is Unknown command: "M20"

### Steps to Reproduce

1. Upload the firmware
2. Receive feedback from the board, not: Unknown command

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Original Prusa

### Electronics

Ramps 1.4

### Your Slicer

Slic3r

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I try the upload with "Auto Build Marlin" and without, the same situation.

ellensp, 2022-09-05:
You haven't enabled SDSUPPORT

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2022-09-05:
The "Your Configuration provides no method to acquire user feedback!" means you have no lcd or other system ebaled to give you any feedback.

closed 2022-09-05 by ellensp

thingAS, 2022-09-05:
Thank you so much!