MarlinFirmware/Marlin #24739: [FR] Pause instead of halt in temperature problem

fsadak, 2022-09-06:
### Describe the feature you want

Wouldn't it make more sense to put the printer into standby instead of halting and resetting the printer due to a bed or nozzle heat problem?  Sometimes, long-term printing is wasted due to the necessity of resetting, although there is a chance to easily solve the problem and continue printing.  This feature can be turned on or off from the configuration.

ellensp, 2022-09-06:
No.  Its a safety feature. 
Printer needs to halt and scream its head off that a person needs to look at it, now.

If a mosfet fails to the on state you need to look at it immediately as heater will just keep heating till power is removed.
The print is already lost 
.

fsadak, 2022-09-06:
Okay, you're right about that and the user should intervene immediately for safety, but what if the problem is just because of an airflow around or because of the geometry of the part cooling fan  airflow hits the part and returns to cool the nozzle?  Such a situation can be easily resolved and the print can continue.  I'm not saying that there are no security measures, but instead of halting, all heating outputs can be turned off and the printer can be put on standby.  If the problem seems unresolvable, then the user can turn off or reset the printer.

ellensp, 2022-09-06:
Such a flawed machine should be scrapped immediately.

fsadak, 2022-09-06:
What kind of flaw?

EvilGremlin, 2022-09-06:
These are problems that must be always solved in hardware.

fsadak, 2022-09-06:
Yes, I can fix some hardware problems when the printer in standby and that's why I want this feature.  Also, maybe one day, like me, you can forget the ceiling fan on and lose 9 hours of print due to the cooling of the bed... :))

EvilGremlin, 2022-09-06:
Ceiling fan is no of no concern to printer firmware, as well as any other user errors of such kind.
Yes, what you want is very easy to do. No, we won't help you make your machine less safe. If you really want it that much - gotta figure it out yourself.

fsadak, 2022-09-06:
The ceiling fan was just an extreme example.  However, I don't think there is a safety difference between turning off all the heating outputs and putting the printer in standby mode and letting it halt.  In addition, with such condescending attitudes, Klipper started to become more popular day by day.

fsadak, 2022-09-06:
In addition, this feature can only be active when the heaters lose heat, and the printer may crash again in an uncontrolled increase...

ellensp, 2022-09-06:
2nd most common failure mode. thermistor moves off of the hotend.
Printer sees a temperature drop or lack of heating...

hotend is still burning down your house

EvilGremlin, 2022-09-06:
This topic was discussed many many times before. The reason its kill and not pause is to nudge users to solve their heating issues properly. Also when you have PSU_CONTROL - it kills main power, which is most desired. When you don't - yeah, not much difference between kill and pause electrically.
By the way, don't think your example scenarios weren't thought of -  configs contain everything necessary to crutch your underpowered bed temp fluctuations.

oliver-eifler, 2022-09-06:
As written above you could deactivate any 'security' feature or modify them but don't blame marlin team or anyone else if something goes terrible wrong. 
BTW: klipper also shut down the mcu if something is wrong with temperatures.

InsanityAutomation, 2022-09-06:
> Okay, you're right about that and the user should intervene immediately for safety, but what if the problem is just because of an airflow around or because of the geometry of the part cooling fan [...]

Thats what Adaptive Fan Slowing resolves. If the part cooler is overpowering the heater on bridging etc, enabling this and increasing the hysteresis window (The default values are too tight for it to function well) will allow the firmware to prevent the error altogether by throttling the part cooling fan. The right path is to catch and prevent errors before they occur if its within the systems control. Then if it still occurs, screaming halt is much better than the 1 in a million printer burned down my house...

https://fortune.com/2022/06/09/alibaba-sued-for-selling-a-3d-printer-that-caused-a-fire-that-killed-a-man/

classicrocker883, 2022-09-12:
@fsadak I get what youre saying, and what you want is to change some values in the firmware or settings like: 

- check your PID tune
- change the hysteriou
- disable `#define THERMAL_PROTECTION_HOTENDS `(highly **_NOT_** recommended!!!)
<br>

- and for the best solution, get a better fan shroud. I have a dual cooling fan shroud with better part cooling fan ducts [at my thingiverse.](https://www.thingiverse.com/thing:5263939)
- or get a better heater like 50w + and a cartridge style thermistor like HT-NTC100k

Tupson444, 2022-12-08:
I completely agree that you can't be too safe. Better to have many false alerts (and some failed prints) and a single prevention of overheating, then to risk a fire (or at least hardware damage).

Turning off the thermal protection is a bad idea, instead it's better to increase THERMAL_PROTECTION_PERIOD and THERMAL_PROTECTION_HYSTERESIS, but not too much. On my printer, I have ADAPTIVE_FAN_SLOWING on, just in case.

For safety (and print quality), it's the best to adjust the hardware and room conditions (temperature, airflow, etc.) in such a way that temperature fluctuations caused by airflow are minimal and leave the default settings, so if thermistor readings happen to be incorrect, the firmware would be able to detect it more quickly, thus lowering fire risk.

Power Loss Recovery can be used to try to save the print after halting (although maybe not for materials that require high bed temperature, as they would pop off the surface when cooled).

--------

> Such a flawed machine should be scrapped immediately.

These kind of replies really aren't helpful. _(A machine is flawed beyond any hope of using if it has a non-optimal fan shroud?)_

Also, here is one example:
When testing a newly built or upgraded printer, one may want to print something and watch it until it finishes to see if everything is set properly. If temperature fluctuations are above acceptable, but still safe, it may be better to let the printer continue printing to learn more. To make this possible, thermal protection can be turned off, which wouldn't be a good idea even for testing (especially if an untested printer is left unattended even for a short time). What @fsadak suggested – pause instead of halting – could be used in this case (and other possible uses), which would give insight when the thermal protection would activate (and still save the day if a heating problem does occur) while letting you continue the print. After testing and adjusting the printer, this would be turned off just in case.

Besides, Marlin doesn't prevent an average user from turning off thermal protection, so having a setting to pause instead of reseting in case of thermal protection (which would be, of course, by default turned off and would have a proper explanation and warning) wouldn't make Marlin any less safe than it is now, so the arguments that are absolutely against it whatsoever are off.

I would add that it may not be important enough to spend time to implement and thoroughly test it (if nobody actually needs it), so that could be the reason to reject the proposal.

masterxq, 2024-08-22:
In my case the temperature can not always be held if very high flow rates are configured and the filament is not dry enough. The maximum power of the heater will no longer be able to meld the filament in the needed speed. The fix is easy: slow down the speed.

Still it is a safety feature. But it will not conflict.
Just add a second max_error threshold for pause. If pause is started and the error counter still counting up the other max_error threshold will be reached and we the printer will halt.

It will not down rate the safety if the printer pauses with error count 90 and halt when error count 120 will be reached. In all cases I had in about 20000 h (reached the limit approximately 10 times) of printing with different printers, pause and slowdown would have fixed the underlying issue (heater has not enough power for the flow rate).

Could be really nice to have this, would have saved me some kg of filament...