MarlinFirmware/Marlin #24745: [BUG] Homing failed on Fysetc Spider2.3 board with bugfix-2.1.x branch

Johnny-Martin, 2022-09-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My board is Fysetc Spider2.3, when I select the "Auto Home" menu item and press the button, the XYZ axises are not homing to  endstop, they just move a little distance towards the endstop and then stoped. 

I used M119 to test the endstop state with X_endtop triggered:
[image]

The resule is as expected.

My configuration for Spider2.3:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9536337/Configuration.zip)

This phenomenon appears every time when I try to home the XYZ axises.

### Expected behavior

I expect the XYZ axises can move and touch the endstop when I press the "Auto Home" menu item.

### Actual behavior

The XYZ axises jost moved a little distance and stoped, homing failed.

### Steps to Reproduce

1. Set the config file as Configuration.zip shows.
2. Build the Marlin project, cand copy firmware.bin to TF card.
3. Reboot the Fystc Spider2.3 board, wait for system start.
4. Select the "Auto Home" on menu, and press.

### Version of Marlin Firmware

Marlin 2.1.1 (Sep  9 2022 22:11:48) 

### Printer model

MB

### Add-ons

TMC2209

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:Homing Failed
Error:Printer halted. kill() called!
Error:Printer halted. kill() called!

ellensp, 2022-09-13:
not enough information...

Are you using physical end stops or sensorless?

If physical, did you notice this board has endstop in a different order than most
X-MAX, Y-MAX, Z-MAX, Z-MIN, Y-MIN, X-MIN

https://wiki.fysetc.com/images/Spider_V2.3_Pinout.jpg

Johnny-Martin, 2022-09-13:
I used physical end stops. And the order is right, if not, the result of M119 will not as expected, right?

Johnny-Martin, 2022-09-21:
I checked the configuration and my wiring again, the problem still exist.
[image]

 The Y axis homing failed. then I removed the Y axis steper,  and start "auto home",  press the Y endstop with my fingre, homing succeed......

Johnny-Martin, 2022-09-29:
Finally, I got it! The root cause is, the "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" micro is defined as default, and I didnt have a probe!

closed 2022-09-29 by thisiskeithb

Johnny-Martin, 2022-09-29:
BTW, I think the "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" micro should not defined as default

ellensp, 2022-09-29:
This does not make sense.

Marlin has 

/**
 * Conditionals based on the type of Bed Probe
 */
#if HAS_BED_PROBE

...Do stuff...

#else
  // Clear probe pin settings when no probe is selected
  #undef Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
  #undef USE_PROBE_FOR_Z_HOMING
#endif