MarlinFirmware/Marlin #24756: [BUG] waiting for user after bed leveling

AimSai59, 2022-09-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I have an Anet A8 under marlin 2.1.1, equipped with a BLTouch sensor and the bed leveling is made thanks to 16 contact points.
I use cura to launch the prints.
After sending G29 for the bed leveling, the printer asks for a user action. I have to press a button to continue printing. Sometimes 1 or 2 more commands come through but I always get stuck at some point.

From what I have read, it seems that the G29 command is too long, the printer not receiving any command for several minutes (16 points to measure is long), blocks the next events. If sometimes 1 or 2 additional commands pass, it is probably because they were stored in the buffer.

This blocking also exists during slow printing of long straight lines, which adds to the thesis of a timeout.

My question is: is it possible to disable this security timer for the time of the G29 or is it possible to extend this delay in Marlin?

### Bug Timeline

Always

### Expected behavior

Printing should continue after bed leveling

### Actual behavior

Printing stops after bed leveling or slow printing of long straight lines

### Version of Marlin Firmware

2.1.1

### Printer model

Anet A8

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-09-12:
No config files attached, please attach bugifix config files that exhibits this issue

renzaaa, 2022-09-12:
Same problem

AimSai59, 2022-09-12:
> No config files attached, please attach bugifix config files that exhibits this issue

uploaded

closed 2022-11-23 by github-actions[bot]