MarlinFirmware/Marlin #24757: [FR] Add support for TriGorilla Pro A V1.0.4 for Anycubic Kobra, Anycubic Kobra Max

tensiondriven, 2022-09-13:
### Is your feature request related to a problem? Please describe.

Need to add support for Trigorilla Pro 1.0.4a

### Are you looking for hardware support?

Yes, the Anycubic Kobra and Anycubic Kobra Max use the TriGorilla Pro A V1.0.4 board, but Anycubic has not published the pinouts for this board.  

### Describe the feature you want

Add support for TriGorilla Pro A V1.0.4 board to support anycubic Kobra, Kobra Max.

### Additional context

https://www.chitusystems.com/ may be the manufacturer of the Trigorilla boards (not confirmed)

Discussions on Reddit:
- https://www.reddit.com/r/anycubic/comments/uz55jx/marlin_on_kobra_max/
- https://www.reddit.com/r/anycubic/comments/wqxquu/a_path_to_klipper_on_the_kobra_kobra_max/

Picture of the Mainboard (Anycubic Kobra) TriGorilla Pro A V1.0.4
https://imgur.com/a/N4fyafO
Main Chip on the board appears to be a HC32F460 Cortex-M4

xC0000005, 2022-09-14:
Chitu is the manufacturer (see the trigorilla_pro_disk environment, where the firmware is encrypted.)  You should be able to start from the pro and work out the pins.

tensiondriven, 2022-09-14:
I'll give it a shot.  Can you provide an overview of the process for working the pin-outs?  I'm technical (software developer), but not terribly experienced with reverse engineering hardware.  I've heard of other people suggesting that this can be done, but need an outline/overview of where to start.   Any links, resources etc would be useful!

xC0000005, 2022-09-14:
First step - Backup your firmware & bootloader. Pull the boot jumper and start that sucker up and download it using ST's flasher. Now you can always go back. Put that boot jumper back on.

Second step-
Get SOMETHING to build for that board. If there's no config, I'd suggest using the AnyCubic Predator config. It's for a delta but it'll let you build something. 
Copy over the config files from the config repo (they're under examples, delta, anycubic, predator IIRC)
You'll want to add a motherboard id to boards.h (don't worry about the ID, but locate it near the other trigorilla one.
Youll want to add a case for that motherboard to mb.h
You'll add a pins file for your rev (copy the existing trigorilla and hope!)
Set your board in configuration.h t your new trigorilla variant.
run pio run -e trigorilla_pro_disk, copy the update.zw file it created to the SD, put it in and turn it on. It should beep a bunch and hopefully boot up.

If it doesn't, start turning off features (TFT display comes to mind).  By default it'll probably want you to listen to usb at 115200, 8, N, 1 and see if it boots up. Sometimes Marlin will tell you what goes wrong. SD_SUPPORT is another thing to turn off if it won't boot.

Most people will use a multimeter with continuity testing and the pinout of the chip to determine endstops, heaters, and stepper pins, filament runout and so on.

If trigorilla_pro_disk won't update for you, mail me (git hub handle at gmail ) the backup you made and I'll find out the offset for the bootloader.

I leave heaters unplugged until verified and I test:
endstops,
Thermistors,
heaters,
steppers,
runout/level/etc

in that order.

tensiondriven, 2022-09-15:
Brilliant, thank you.  That all made sense to me.  Hopefully I won't fry the thing in the process.  A couple of Q's:

What is **ST's Flasher**?  Presumably this is the flashing utility for the STM board, but wanted to check.  Hoping it runs on linux or mac, though i guess worst case I can get windows running on something.

What is `pio`?  Is this a binary that comes with the flasher?

When you say
> If trigorilla_pro_disk won't update for you
Are you saying that if I'm unable to boot the board with using the trigorilla firmware binary from the Marlin repo then you can lend a hand?  Just trying to grok "won't update".

This is most excellent, thank you for effort - i appreciate it.

xC0000005, 2022-09-15:
What processor is that on the board? A GD STM clone or actual ST? Regardless, you want to find the pinout for chip and look for boot0 and trace it, see if it goes to a pad. If it does, pulling it high (any 3.3v source) and then powering on will invoke the serial bootloader.

https://www.youtube.com/watch?v=vL-JXBVUNRg shows how to back up chitu boards (which this is). I don't know if ST's tools will read GD devices. If you succeed in saving it off, drop a copy of it to me in email and I'll see if it has any different encryption than the other chitu boards. (If there's a firmware update already available for it, point me to it).

xC0000005, 2022-09-15:
Hey, sweet - there's no encryption on the firmware now. That's a great sign.

xC0000005, 2022-09-15:
This is not a chitu board (it's made by them but doesn't use their encryption at all). I'd see if there's any other boards based on that hardware and start from those. Still use the multimeter to map the pins, but it's not exactly a chitu/trigorilla derivative like you'd expect. So the real question is "What board do you start from as a basis?" and that depends on the MCU.  Sorry for the bad information on what environment to use.

tensiondriven, 2022-09-15:
No problem, still super thankful for the guidance.  

I also mis-spoke; the MCU on the board is the `HC32F460` (which apparently only recently received support for Klipper)

Yes, the firmware is a modified Marlin.  No, they didn't release the source (and yes, they're in violation of GPL.  There are a few people on the internet who are upset about that.)

Out of curiosity, how did you determine that there's no encryption on the firmware? 

Knowing that the board is based on `HC32F460`, does that help you advise me on what base firmware / board to start from as a basis?

thisiskeithb, 2022-09-15:
HC32F460 is not supported in mainline Marlin yet, so they *really* need to post or provide their source code to see what was done to support this MCU.

ST’s licensing does not allow running some of their code on non-ST MCUs, so we’d need to verify some things before adding support (see: GD32 support).

xC0000005, 2022-09-15:
How did I know there's no encryption? I opened in in binary ninja and read the vector table, guessed a few values and got a relatively decent disassembly. Like thisiskeithb, I'd love to see what they did. Looks to me like it would be a different HAL (and the timers are...limited).

xC0000005, 2022-09-15:
https://github.com/wellrun/hc32f460/tree/master/hc32f460_sdk/%E7%BC%96%E7%A8%8B%E5%B7%A5%E5%85%B7 has the programmer, so you could (maybe) backup the firmware and bootloader, if you can translate it.

tensiondriven, 2022-09-20:
> Bonjour Si ça peut aider j'ai trouvé quelques pin pour kobra max voir photo
> Translated: Hello If it can help I found some pins for kobra max see photo

Fantastic!  This is a huge step forward.. I'm afraid I don't know what all of the names for the findings mean so I may need some help making this actionable

xC0000005, 2022-09-20:
> 

If you own a cheap dev board for this, it should be possible to cobble together enough Arduino functions on top of the MDK/Pack files from the website I linked to. This thing will be a lot like the M200 series of printers, motion controller on a limited MCU and probably a smart LDC (I'd guess).

tensiondriven, 2022-09-21:
@xC0000005 There's enough here that I would need to learn to pull this off that I expect I won't have time/energy to do it for some time.  I'll try posting a link to this thread on the Anycubic Reddit and maybe someone will be interested in coordinating with me on moving it forward.

Thanks x100 for everything so far!

ServerOnTheMove, 2022-10-12:
Seems that Anycubic published their development(?) branch of the Max firmware. Maybe this will help?

https://github.com/ANYCUBIC-3D/Kobra_Max

MaxWinterstein, 2022-11-05:
> Seems that Anycubic published their development(?) branch of the Max firmware. Maybe this will help?
> 
> https://github.com/ANYCUBIC-3D/Kobra_Max

And also for Kobra:
https://github.com/ANYCUBIC-3D/Kobra

ellensp, 2022-11-06:
Yes the source is out, unfortunately you cannot use platformio, it is designed for Keil uVision

marlomgirardi, 2023-01-01:
I believe the Ender 2 Pro has the same MCU as this board https://github.com/MarlinFirmware/Marlin/issues/24338 as it might not work with the current STM32

JoyceKimberly, 2023-03-31:
I've been working on updating the Kobra firmware and I have managed almost all except HAL, but now I'm at the limit of my knowledge and it really needs some proper programmers to go over it, which I'm hoping hang around here :)
https://github.com/JoyceKimberly/Kobra-updated

maikramer, 2023-07-19:
For the people with no hope, well, I'm almost there.
I built a firmware from marlin upstream. It can be built with platformio, like normal marlin. I made it changing almost nothing from the marlin source, only adding code in a separate repo for the Arduino related code. Well, for someone wanting to take a look. Filament sensor is not stable yet, with a feel things.

The repo is https://github.com/maikramer/MarlinKobraMax

I work on it everyday, and I'm keeping marlin updated.

ntrrgc, 2023-08-08:
> Yes, the firmware is a modified Marlin. No, they didn't release the source (and yes, they're in violation of GPL. There are a few people on the internet who are upset about that.)

About a month later, the firmware source code was finally made available in their GitHub:

https://github.com/ANYCUBIC-3D/Kobra
https://github.com/ANYCUBIC-3D/Kobra_Max

There is this reddit post that explains how to build it in Windows using Keil https://www.reddit.com/r/anycubic/comments/y2waxu/tutorial_how_to_build_anycubic_marlin_source_code/ I couldn't find info myself about how to get it working in something more open, like PlatformIO, but seeing recent Klipper work on that specific MCU chip makes me hopeful: https://github.com/Klipper3d/klipper/pull/5208

The source code has personally been useful when troubleshooting problems with the printer, like it shipping with a different thermistor that they configured in the firmware :sweat: I swapped the thermistor for the one in that source code and was able to finally get PID not being a wild oscillating mess after auto-tuning!

I found it weird that in Configuration.h you can see `#define CUSTOM_MACHINE_NAME "Anycubic Viper"` (instead of Anycubic Kobra), but I could confirm that indeed that's how my Kobra with stock firmware identifies over serial:

> FIRMWARE_NAME:Marlin bugfix-2.0.x (Dec 22 2021 14:30:26) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic Viper EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

Tannoo, 2023-09-05:
I have recently got possession of the Kobra Max. Well, this has been interesting. I have been able to compile all of the official published codes, but they do not run on the printer. I am still digging to get something working via PlatformIO before I throw out this useless control board.

Tannoo, 2023-09-05:
> For the people with no hope, well, I'm almost there. I built a firmware from marlin upstream. It can be built with platformio, like normal marlin. I made it changing almost nothing from the marlin [...]

I am having issues with getting platformIO to get with the ball game.

Platform Manager: Installing git+https://github.com/maikramer/platform-hc32f46x.git
UserSideException: Please install Git client from https://git-scm.com/downloads

I have all forms of GIT installed and cloned that repo. What does it want from me?

Tannoo, 2023-09-05:
> I am having issues with getting platformIO to get with the ball game.
>
> Platform Manager: Installing git+https://github.com/maikramer/platform-hc32f46x.git UserSideException: Please install Git [...]

Nevermind.... I had to restart VSCode.

Tannoo, 2023-09-05:
Still cannot get it to compile. There is a reference to a "loop()" in main.cpp of the platformIO toolchain that is says doesn't exist.

Tannoo, 2023-09-05:
```
    extern void timer01B_init(); // used for beep duration timer
    extern void timer02B_init(); // soft serial
    extern void timer41_init();  // 1k Hz, used for temperature tick
    extern void timer42_init();  // step motor

    extern void setup();
    core_hook_post_setup();

    // call loop() forever
    while (1) {
        core_hook_loop();
        extern void loop();
    }
    return 0;
}
```
I added "extern void " to the 'loop()', 'setup()', 'timer01B_init()', 'timer02B_init()', 'timer41_init()', and 'timer42_init()'.

Now, it compiles.

Let's see if it runs.

Tannoo, 2023-09-05:
No go. The printer beeped several times before the screen booted up. The system just hangs after that. Restarting it does no good.
Put stock firmware back on and it is running again.

ellensp, 2024-10-09:
I took a good look into this

Here are the issues I identified with just supporting this motherboard (there are more issues with additional peripherals, ie the display and the loadcell) 

These features are not in the current HC32 HAL

Uart4  is not supported (this seems easy to add)  Is required for the LCD serial port
No arduino tone support for onboard buzzer
The board does not have a real eeprom, it is using the end of flash as a eeprom storage, but FLASH_EEPROM_EMULATION is not supported. 
SPI.h, is required for tmcstepper library, even when the tmc2209 and tmc2208 stepper drivers don't need this, but SPI.h it not implemented (I got this part to compile by simply creating a some stub code)

@shadow578 Any chance you could add some or all of these to the hc32 HAL?

shadow578, 2024-10-09:
Hi @ellensp 

> Uart4 is not supported (this seems easy to add) Is required for the LCD serial port

AFAIK the uart peripherals of the HC32F460 are not fixed to any pins, so just replacing UART4 with any of UART1-3 while keeping the pins should work just fine. 
If not, adding UART4 support should be fairly easy.

>  No arduino tone support for onboard buzzer

really not sure how the tone stuff would've to be implemented, but there's a PWM library in the core already. Maybe someone can take a look?

>  The board does not have a real eeprom, it is using the end of flash as a eeprom storage, but FLASH_EEPROM_EMULATION is not supported.

it'd probably be easier to just use the SD card as a eeprom replacement, idk how long the flash would hold up to repeated reprogramming (assuming that this cheap-ish chinese MCU uses similarily cheap-ish chinese flash)...

>  SPI.h, is required for tmcstepper library, even when the tmc2209 and tmc2208 stepper drivers don't need this, but SPI.h it not implemented (I got this part to compile by simply creating a some stub code)

stub headers for the arduino SPI library are already present at https://github.com/shadow578/framework-arduino-hc32f46x/tree/main/libraries/SPI.

as for the implementation, take a look at the DDL examples at https://github.com/shadow578/hc32f460-documentation-and-sdk/tree/main/sdk/Driver%20library%20and%20samples/hc32f460_ddl_Rev2.1.0/example/ev_hc32f460_lqfp100_v2/spi, especially the `*_polling` and `*_interrupt` examples.
for the integration in the arduino core, especially interrupt assignment, the [`Usart`](https://github.com/shadow578/framework-arduino-hc32f46x/blob/main/cores/arduino/drivers/usart/Usart.cpp) class should be similar enough to use for inspiration.

since my printer doesn't use SPI, and dev boards for the HC32F460 are basically unobtainium, i can't really do much since i cannot test anything. 
but if anyone did open a PR to add SPI, i'd be happy to include it.

---

a bit more overall: the HC32 HAL is written in such a way that, to add support for a new board, only the pinout has to be added (assuming that no missing libraries - like SPI - are needed).
changes to anything but the arduino core and HAL will __never__ be needed.

for debugging + programming, the arduino core supports both via a SWD probe (like the picoprobe).

---

if there's any questions regarding the HC32F460, i'd be happy to help. 
Tho right now, i'm working on my Bachelor's thesis, so i don't have too much time on my hands for testing.

ellensp, 2024-10-09:
you can actually get these boards from aliexpress https://www.aliexpress.com/item/1005004178328026.html

shadow578, 2024-10-09:
> since my printer doesn't use SPI, and dev boards for the HC32F460 are basically unobtainium, i can't really do much since i cannot test anything.
but if anyone did open a PR to add SPI, i'd be happy to include it.

actually, after looking at the SPI ddl, it seems to be fairly straightforward to implement the basic functions of the library.

if someone'd like to test, see https://github.com/shadow578/framework-arduino-hc32f46x/pull/29.
implemented it according to the 3-wire polling example @ https://github.com/shadow578/hc32f460-documentation-and-sdk/blob/main/sdk/Driver%20library%20and%20samples/hc32f460_ddl_Rev2.1.0/example/ev_hc32f460_lqfp100_v2/spi/spi_three_wire_tx_and_rx_polling/source/main.c.

since i currently don't have access to a environment where i could compile this, it's very likely that the lib fails to compile.

emiliogc673-sketch, 2025-10-28:
In this photo, the connector marked in red has three cables missing. I don't know where they go or how they're used: LEVE, SDA, and SCL.

I'd appreciate it if you could help me.And in the connector photo it is on the plate that I have above in the header already marking the cables that I have located, those 3 leve, sda and scl would be missingEnviado desde mi Galaxy
-------- Mensaje original --------De: ellensp ***@***.***> Fecha: 28/10/25  11:28  (GMT+01:00) Para: MarlinFirmware/Marlin ***@***.***> Cc: emiliogc673-sketch ***@***.***>, Mention ***@***.***> Asunto: Re: [MarlinFirmware/Marlin] [FR] Add support for TriGorilla Pro A V1.0.4 for Anycubic Kobra, Anycubic Kobra Max (Issue #24757) ellensp left a comment (MarlinFirmware/Marlin#24757)
Anycubic-Vyper-Extruder-Board-P010115-26721.png (view on web)
Anycubic-Vyper-Extruder-Board-P010115-26721_2.png (view on web)
You can work out most of them from the extruder board images
Firstly the 4 stepper motor pins OA2,OA1,OB1 and OB2  are not on the extruder board and neither are the filiment runout pins f_out and a gnd. so that leaves 16 pins ie the rest all go to the extruder board
But I don't have a  22pin - 16pin wiring diagram.

Wepon-X, 2026-11-13:
> Brilliant, thank you. That all made sense to me. Hopefully I won't fry the thing in the process. A couple of Q's:
>
> What is **ST's Flasher**? Presumably this is the flashing utility for the STM [...]

Did you extract the copy and save the original firmware of Trigorilla Pro A 1.0.4, can you share the dump file. Actually I am facing a board burn out. The original chip got short circuit, so I bought a new chip and soldered on board, but since my chip is empty, it need anycubic bootloader  first to boot the firmware. If anyone who has anycubic kobra 1 printer, he can extract the copy of bootloader + firmware from its Tri-Gorilla Pro A 1.0.4 board using a J-link segger device or ST link, both work fine for this work. You just need to hookup the P4 connector with the device start exracting the copy.

ellensp, 2026-11-13:
@Wepon-X 
I have the AnyCubic Kobra Plus, same TriGorilla Pro A V1.0.4 motherboard, so hopefully it has the same bootloader.

After a bit of a battle (it didn't want to give it up!, but wasn't locked)  

Here is a  512k flash, ota (One-Time Programmable), and even EFM (Embedded Flash Memory) dumps

[kobra_plus_backup.zip](https://github.com/user-attachments/files/34796134/kobra_plus_backup.zip)

Bootloader is from 0x00000000 - 0x00007FFF

Wepon-X, 2026-11-13:
> [@Wepon-X](https://github.com/Wepon-X) I have the AnyCubic Kobra Plus, same TriGorilla Pro A V1.0.4 motherboard, so hopefully it has the same bootloader.
> 
> After a bit of a battle (it didn't want to give it up!, but wasn't locked)
> 
> Here is a 256k flash, OTP (One-Time Programmable), and even EFM (Embedded Flash Memory) dumps
> 
> [kobra_plus_backup.zip](https://github.com/user-attachments/files/34797726/kobra_plus_backup.zip)
> 
> Bootloader is from 0x0000 - 0x7FFF
> 
> EEPROM is virtual at the end of flash 0x3E000 onwards (do not know if that is stock; my machine is second-hand, and I don't know if it's been rewritten)

Thanks a lot for your help, brother. Even though your printer was a Kobra Plus, but boot loader was same due to same board type. So I was successfully able to extract the first 32KB bootloader file and merge it with the official firmware. It booted beautifully. Thanks once again.

Wepon-X, 2026-11-13:
I am attaching the firmware file that contains both the official bootloader and firmware, merged for the Anycubic Kobra 1 3d printer. Anyone who needs it can use it without any doubt. It should boot beautifully via SD card, but if it doesn't, due to the new chip or any other reasons, I prefer flashing the firmware via ST-Link and J-Link Segger V9.

[Anycubic_Kobra_1_Official bootloader.zip](https://github.com/user-attachments/files/34830250/Anycubic_Kobra_1_Official.bootloader.zip)