MarlinFirmware/Marlin #24758: [BUG] Tool change runs nozzle into print due to applying HOTEND_OFFSET_Z before EVENT_GCODE_TOOLCHANGE is run

ansonl, 2022-09-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When executing a toolchange, Marlin will adjust the z position of the bed to account for HOTEND_OFFSET_Z before EVENT_GCODE_TOOLCHANGE_T is run. In a dual nozzle setup with any Z offset between the nozzles, this will move the current (switching away from) nozzle into the printed object if the active nozzle has a lower Z offset than the newly active nozzle. Code in EVENT_GCODE_TOOLCHANGE_T will move the lower nozzle upwards out of the way so that the newly active nozzle is the lowest nozzle. 

Thus EVENT_GCODE_TOOLCHANGE_T should run before Marlin toolchange adjusts the bed position to account for HOTEND_OFFSET_Z. The user can add gcode at the beginning and end of EVENT_GCODE_TOOLCHANGE_T to lower and raise the bed to avoid any nozzle crash while performing the toolchange. Only when the toolchange is complete after each setup-specific EVENT_GCODE_TOOLCHANGE_T should the bed be offset to account for HOTEND_OFFSET_Z. 

Before toolchange from T1 to T0:
```
                   |T0|     
new active nozzle =>▼    |T1|
                          ▼ <- currently active nozzle      ______print_surface_______
```

Bed adjusted for HOTEND_OFFSET_Z before T1 gets raised by printer specific EVENT_GCODE_TOOLCHANGE_T 
```
                   |T0|     
new active nozzle =>▼    |T1|                                ______print_surface_______
                          ▼ <- currently active nozzle now below print surface for a second between HOTEND_OFFSET_Z adjustment and execution of EVENT_GCODE_TOOLCHANGE_T 
```

It is not safe to adjust for HOTEND_OFFSET_Z before running EVENT_GCODE_TOOLCHANGE due to the active nozzle offset. 

### Bug Timeline

Old issue since HOTEND_OFFSET_Z and EVENT_GCODE_TOOLCHANGE were used togther

### Expected behavior

Hotend offset adjusted after EVENT_GCODE_TOOLCHANGE which raises the lowers nozzles is executed. 

### Actual behavior

Hotend offset adjusted before EVENT_GCODE_TOOLCHANGE and the nozzle crashes into the printed object. 

### Steps to Reproduce

1. Set HOTEND_OFFSET_Z non zero value such as { 0.0, -5.00 }
2.  `#define EVENT_GCODE_TOOLCHANGE_T0 "G91\nG0 Z3 F2400\n printer nozzle adjustment movement here \nG91\nG0 Z-3 F2400\nG90"`
3. Switch to T1
4. Switch to T0
5. Nozzle will be raised by Z_RAISE amount (default 2mm), then lowered by 7mm, before running EVENT_GCODE_TOOLCHANGE_T0 . T1 nozzle will be below the printed object surface height by 7mm before running EVENT_GCODE_TOOLCHANGE_T0. 

### Version of Marlin Firmware

2.1.1 bugfix

### Printer model

Ultimaker Original+ DXU

### Electronics

Stock ultimainboard v2.1.4

### Add-ons

DXU dual extrusion with mechanical switching nozzle

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration uses switching extruder enhancement #24553 currently awaiting merge
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9553107/Configuration.zip)

DerAndere1, 2022-09-13:
Does it help to increase `TOOLCHANGE_ZRAISE` (by editing configuration.h or by using M217 Z...) ?

ansonl, 2022-09-13:
@DerAndere1 increasing z_raise does not work because the nozzle gets raised by z_raise amount then lowered right afterwards to account for hotend offsets which account for the z_raise height. So it is a quick z_raise up followed by a down movement which is negative z_raise height and hotend offset together so the newly active nozzle is aligned with the print surface (before tool change Goode is run)

DerAndere1, 2022-09-13:
I proposed a possible fix in the comment section of pull request 24553

ansonl, 2022-09-14:
I used your suggestion in the comment to create my fix [#3](https://github.com/EvilGremlin/Marlin/pull/3) for #24553

ansonl, 2022-11-16:
Replying to keep this issue open until fix in #24553 is merged (the PR is ready to be merged). @thinkyhead @EvilGremlin

thisiskeithb, 2022-11-16:
Since there already a PR open with a fix, this can be closed.

closed 2022-11-16 by thisiskeithb