MarlinFirmware/Marlin #24762: [BUG] Z3 stepper does not move with `do_blocking_move_to_z`

USA-RedDragon, 2022-09-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using the following for Z axes:

```
|   3   |
|       |
| 1   2 |
```

I noticed some behavior where G34 would converge Z1 and Z2 appropriately, but Z3 would always be consistently offset and not move, no matter the number of iterations. As an example of G34 outputs:

```
Z2-Z1=7.29 Z3-Z2=7.37 Z3-Z1=14.65
Z2-Z1=0.72 Z3-Z2=13.93 Z3-Z1=14.65
Z2-Z1=0.62 Z3-Z2=14.48 Z3-Z1=13.86
Z2-Z1=0.03 Z3-Z2=13.89 Z3-Z1=13.86
Z2-Z1=0.02 Z3-Z2=13.87 Z3-Z1=13.86
Z2-Z1=0.01 Z3-Z2=13.82 Z3-Z1=13.81
Z2-Z1=0.00 Z3-Z2=13.82 Z3-Z1=13.82
Z2-Z1=0.02 Z3-Z2=13.85 Z3-Z1=13.83
```

After attempting to move the leadscrew of Z3 closer to level by eye and trying again, the lack of convergence became more obvious:

```
Z2-Z1=0.01 Z3-Z2=1.41 Z3-Z1=1.42
Z2-Z1=0.01 Z3-Z2=1.41 Z3-Z1=1.40
Z2-Z1=0.00 Z3-Z2=1.40 Z3-Z1=1.40
Z2-Z1=0.00 Z3-Z2=1.41 Z3-Z1=1.40
Z2-Z1=0.00 Z3-Z2=1.40 Z3-Z1=1.40
```

I enabled `DEBUG_LEVELING_FEATURE` and turned it on with `M111 S32` and verified that the correct offset was being requested.

I also wanted to ensure that my Z1-Z3 mapping was correct, so I wrote a new G-Code command (M130, I can PR it if anyone might find this feature useful) to move a single Z axis at a time. 

I grabbed the stepper index from `Z` and the distance from D:

```
if (parser.seenval('Z')) stepper_index = parser.value_int();
if (parser.seenval('D')) stepper_move = parser.value_int();
```

Then separate the axes in the stepper module:

```
stepper.set_separate_multi_axis(true);
```

Locked all steppers except the selected one:

```
stepper.set_all_z_lock(true, stepper_index-1);
```

Finally, I do the Z move:

```
do_blocking_move_to_z(stepper_move + current_position.z);
```

Using this command from OctoPrint's terminal, such as `G130 Z1 D10` moves the Z axis +10mm. This worked as expected for Z1 and Z2, but Z3 completely fails to move. I believe this is why G34 is able to converge Z1 and Z2 while Z3 stays consistent.

### Bug Timeline

First setup of a new printer

### Expected behavior

I expect Z3 to be able to move with separated axes enabled.

### Actual behavior

Z3 will not move

### Steps to Reproduce

See the very detailed description. 

### Version of Marlin Firmware

Commit c5af4354498b9c634b397aecfc8034793b724e90

### Printer model

RatRig VCore 3.1

### Electronics

BTT Octopus V1.1

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9563398/Configuration.zip)

[M130.cpp](https://github.com/MarlinFirmware/Marlin/files/9563405/M130.cpp.txt)

USA-RedDragon, 2022-09-14:
[g34_log.txt](https://github.com/MarlinFirmware/Marlin/files/9563486/g34_log.txt)

ellensp, 2022-09-14:
Where do you have z3 plugged in as most people get this wrong on the octopus..

And did you notice?
`#warning "Note: Auto-assigned Z3 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
`
X  motor 0 
Y  motor 1
Z  motor 2-1 and motor 2-2 (2 plugs one stepper driver)
Z2 motor 3
E0 motor 4
Z3 motor 5  (first unused extruder reallocated as per warning above)

USA-RedDragon, 2022-09-14:
I had it 

`X Y Z Z2 Z3 E0 empty empty empty`

I changed it as you suggested:
`X Y Z empty Z2 E0 Z3 empty empty`

Unfortunately, the printer kills itself in this config,

```
Recv: Z2 driver error detected: 0xC01D0020
Recv: coil short circuit
Recv: echo:Driver error
Recv: Error:Printer halted. kill() called!
```

USA-RedDragon, 2022-09-14:
That was it. Hopefully anyone going through similar issues should be able to find this issue

closed 2022-09-14 by USA-RedDragon