MarlinFirmware/Marlin #24763: 12864kba_v1.2 encoder rotation and sd

mahdiabedi75, 2022-09-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

LCD works. But when you insert the SD card inside it gives 'media init failed 'error. 
also encoder works when clicked it but when you turn it the value goes up once then goes down on the next pulse regardless of direction. 

Have anything else attached to board yet. Just the LCD. 

### Bug Timeline

from start

### Expected behavior

Should move on menu and read the SD card

### Actual behavior

can't read the SD card and rotate value only goes up once, then goes to zero with next pulse.

### Steps to Reproduce

Just upload marlin with platform Io with LCD config on reprap full graphic controller and SD support. 
[marlin.docx](https://github.com/MarlinFirmware/Marlin/files/9568359/marlin.docx)

### Version of Marlin Firmware

bug fix 2.1

### Printer model

custom

### Electronics

ardiano adk + ramps 1.4

### Bed Leveling

No Bed Leveling

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-09-15:
No configs

attached file Marlin.docx is 0 bytes long.

ellensp, 2022-09-15:
Never heard of a 12864kba_v1.2  and google is also coming up blank.  Url where you got it from?

ellensp, 2022-09-15:
The Arduino ADK is also not 100% compatible with the Arduino mega2560..  having and extra spi usb port device added and using D7 for that device.

mahdiabedi75, 2022-09-15:
I see. It most likely the difference in pins. 
I will map the pins and try to match them with the program.

mahdiabedi75, 2022-09-16:
What ever it is. it is on exp2. Cause the LCD and encoder click is on exp1, and it's working. But rotation and SD card is on exp2 are having problems.

mahdiabedi75, 2022-09-17:
welp. found the problem. it's the speed. Whether the bud rate or encoder pulse setting. just have to mess with it until find the right combo.

mahdiabedi75, 2022-09-18:
I feel like this getting ignored somehow. 
but whatever.
After lowering the bud rate, the SD card does not give error anymore. 
but the encoder remains the same.  And the problem is like the number is limited to 1 and 0 just goes back and forth
When the encoder pules setting both is on 1. And when it's other than 1, encoder rotation freeze completely

closed 2022-11-29 by github-actions[bot]

gnfpt, 2023-01-14:
Hello, 

I have the same issue with Aimsoar 12864KBA_V1.2 and Anet mainboard v1.7. I'm using Marlin 2.0.9.5.
I didn't try the SD Card yet but the rotary knob (LS1) doesn't work when rotating, only when clicked.
Also, the button left to the speaker (K1) also doesn't seem to work. The speaker (BZ1) works fine.
I'm not using bed leveling at this time. The only thing attached to the board is the LCD (through as breadboard).

[image]
[image]
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10417051/Configuration.zip)

ellensp, 2023-01-14:
The button not working does not make sense.  
Its a reset button, it should be directly connected to a reset io line. when the switch is closed the reset line is pulled low and the board resets,. no firmware, no settings, nothing firmware related at all

I suspect your exp2 is not wired correctly..
reset is on exp2
btn_enc1 and 2 (ie left and right) are also on exp2

gnfpt, 2023-01-14:
Hi,

I've just confirmed that EXP2.8 is indeed the reset.
Maybe it's not working because the Anet board already has a reset 
button. I'll have to check that.

Thank you,
Gonçalo Farias

gnfpt, 2023-01-14:
Hi,

You were right. After some testing, I've figured out that I had my cable 
incorrectly wired.
After wiring EXP2 correctly both the reset button and rotary knob were 
working.

Thank you,
Gonçalo Farias