MarlinFirmware/Marlin #24764: [BUG] TMC hybrid threshold

cryptoAlgorithm, 2022-09-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hybrid threshold to switch from StealthChop to SpreadCycle persists across resets correctly, but doesn't get written to the TMC driver's registers on reset. This leads to skipped steps initially, which can be fixed by running `M913 X<value> Y<value>` again, presumably to force a write to the driver's registers.

### Expected behavior

Hybrid threshold setting, both hardcoded and thru `M913` should persist across resets _and_ be set to the driver every time a driver is reenabled, on power on and when the PSU is turned on.

### Actual behavior

Hybrid threshold doesn't appear to be written to driver's registers (properly), the driver acts as though no hybrid threshold is set on power on or reset. 

### Steps to Reproduce

1. Set a custom hybrid threshold speed with `M913 X<value> Y<value>` (in my case it was 200mm/s for both)
2. Run `M500` to store values
3. Verify hybrid threshold is applied by moving motors at various speeds
4. Reset controller
5. Move motors at various speeds and observe hybrid threshold no longer applies (motors skip at high speeds, no buzzing sound thats present with SpreadCycle)
6. Run `M913` to verify values are retrieved from EEPROM correctly (they are)
7. Run `M913 X200 Y200` again
8. Move motors at various speeds again and observe hybrid threshold is once again applied

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

TMC2209 driver, ESP32 running marlin

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/9572685/configs.zip)

thisiskeithb, 2022-09-15:
Related bug: https://github.com/MarlinFirmware/Marlin/issues/19050

cryptoAlgorithm, 2022-09-15:
Seems related, but in my case I'm trying to get the opposite to work: I would like my settings to apply after a power cycle, but that doesn't seem to be happening

cryptoAlgorithm, 2022-11-17:
This is still an issue - no fixes have been made to mitigate this.

closed 2023-01-29 by github-actions[bot]