MarlinFirmware/Marlin #24769: [BUG] #24737 causes printer moves to become very slow

rondlh, 2022-09-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Printer moves become very slow (X, Y and Z), I can visually see the deceleration happening.

### Bug Timeline

20202 September 14th

### Expected behavior

Deceleration as before

### Actual behavior

Deceleration are very slow, I'm not sure about acceleration, but they seem fine.

### Steps to Reproduce

1. Do a bed probing, then the slowdown becomes very obvious

### Version of Marlin Firmware

Latest bugfix (September 16th 2022)

### Printer model

Custom

### Electronics

MKS SGEN_L V2.0 (LPC1769)

### Add-ons

BTT TFT35 V3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Host is the BTT TFT35 V3.

I use TMC2xxx drivers, and have setup the stepper parameters accordingly:
#define MINIMUM_STEPPER_POST_DIR_DELAY 20
#define MINIMUM_STEPPER_PRE_DIR_DELAY  20
#define MINIMUM_STEPPER_PULSE 0
#define MAXIMUM_STEPPER_RATE 5000000

I managed to trace back this issue to the code change in planner.cpp #24737
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/9590566/Configuration_adv.zip)

XDA-Bam, 2022-09-17:
Please upload your `Configuration.h` and `Configuration_adv.h`.

rondlh, 2022-09-17:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/9592489/Configuration_adv.zip)

Check the declaration of "inverse_accel", it is declared twice (around line 802 in planner.cpp), once outside the "if" and once inside. The value calculated inside the "if" will revert back to 0.0f after the if statement, causing issues for user that have S_CURVE_ACCELERATION enabled, which uses the value of "inverse_accel".

thisiskeithb, 2022-09-18:
This should now be fixed in https://github.com/MarlinFirmware/Marlin/commit/1e525c597699536d3185dfc92b3066bc6b4ef695

closed 2022-09-18 by thisiskeithb