MarlinFirmware/Marlin #24771: [FR] Support Fysetc Spider2.3 motherboard

Johnny-Martin, 2022-09-17:
### Is your feature request related to a problem? Please describe.

I have a motherboard of Spider2.3, and recently I configure Marlin firmware with BOARD_FYSETC_SPIDER_V2_2，which is not compatible with Spider2.3, like the homing function always failed, and I describe this issue in [here](https://github.com/MarlinFirmware/Marlin/issues/24745)

``

### Are you looking for hardware support?

[Fysetc Spider 2.3](https://github.com/FYSETC/FYSETC-SPIDER)

### Describe the feature you want

Support for Spider2.3

EvilGremlin, 2022-09-17:
quick glance say that at least all XYZ pinouts are exactly same, if it fail homing -  you doing someting wrong

ellensp, 2022-09-17:
agreed, boards are identical.

closed 2022-09-17 by ellensp

Johnny-Martin, 2022-09-17:
I checked my configuration and the wiring several times, didnt figure out any exception, if there are no pinout changes for Spider2.3, please close this issue, I will check my project again, thanks.