MarlinFirmware/Marlin #24772: [BUG] not compiling with SDSORT_DYNAMIC_RAM enabled

Hasi123, 2022-09-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If you enable SDSORT_DYNAMIC_RAM with the latest Marlin, I get this error on compilation:

`C:\Users\David\Documents\GitHub\Marlin\Marlin\src\sd\cardreader.cpp: In static member function 'static void CardReader::presort()':
C:\Users\David\Documents\GitHub\Marlin\Marlin\src\sd\cardreader.cpp:1062:72: error: incompatible types in assignment of 'char*' to 'char [66]'
       #define SET_SORTNAME(I) (sortnames[I] = strdup(longest_filename()))
                                                                        ^
C:\Users\David\Documents\GitHub\Marlin\Marlin\src\sd\cardreader.cpp:1155:13: note: in expansion of macro 'SET_SORTNAME'
             SET_SORTNAME(i);
             ^~~~~~~~~~~~
exit status 1
Error compiling for board Sanguino.
`

Obviously you have to disable SDCARD_RATHERRECENTFIRST and enable SDCARD_SORT_ALPHA and SDSORT_USES_RAM...

### Version of Marlin Firmware

2.1.1

### Printer model

Custom i3

### Electronics

Melzi

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9591848/Configuration.zip)

TjeerdvanDijk, 2022-11-03:
I have this same bug aswell

scrith, 2023-01-30:
I am experiencing this same issue with the release build of 2.1.2 as well.

Bob-the-Kuhn, 2023-01-30:
SDSORT_DYNAMIC_RAM requires SDSORT_CACHE_NAMES to be set to true.  The default configuration_adv.h has SDSORT_CACHE_NAMES set to false.

I'll add a sanity check to warn users of this requirement.

Bob-the-Kuhn, 2023-01-30:
This issue is being closed because it is addressed by PR #25318

closed 2023-01-30 by Bob-the-Kuhn

scrith, 2023-01-30:
> SDSORT_DYNAMIC_RAM requires SDSORT_CACHE_NAMES to be set to true. The default configuration_adv.h has SDSORT_CACHE_NAMES set to false.
> 
> I'll add a sanity check to warn users of this requirement.

How can the two lines labelled "most expensive" and "least expensive" both be required simultaneously?
Also the compiler error says that SDSORT_USES_RAM is required.
Are all 3 options required to be set to true?
How does one reduce the expense of SD sorting if the most expensive option is required to be enabled?