MarlinFirmware/Marlin #24776: [FR] Manual probe. Use saved z value

cfytrok, 2022-09-19:
### Describe the feature you want

There is an option MANUAL_PROBE_START_Z. Comment out to use the last-measured height. How about using previously saved mesh poins as start Z?

borland1, 2022-09-19:
Not sure what your saying would make it easier?  Are you referring to this code snippet in `Configuration.h` ?
```
1910  #if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL, PROBE_MANUALLY)
1911   // Set a height for the start of manual adjustment
1912  #define MANUAL_PROBE_START_Z 0.2  // (mm) Comment out to use the last-measured height
1913  #endif
```
The default is to start manual bed leveling grid at 0.2mm above the `Home` position (typically at (0,0,0).  If you "comment out" line 1912, what are the affects, i.e., what exactly is the "last-measured height"?  It's not clear to me what happens.

After running the Probe Manually to level the bed, are you using the `Edit Mesh` menu feature along with a single layer grid test print?  I have found that 3D printing the test grid and then editing the mesh grid, to be the most accurate way to dial in the mesh.

cfytrok, 2022-10-01:
Yes, I reffering to this this. code.https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/Configuration.h#L1911-L1914
Now there are 2 options: 1) use MANUAL_PROBE_START_Z height as start Z position for everty grid point. (default) 2) use Z-height = 0.4 for first point and Z-height = Z-height of previous point as start Z for next point (comment out 1912 line).
I suggest adding a new option:
3) Load mesh from memory and use mesh Z value ​​as start Z height individualy for every point.

```Edit Mesh``` menu only allows you to replace the value in memory. It does not allow you to check how far the nozzle is from the table after updating the value.