MarlinFirmware/Marlin #24778: No config files for generic Delta all are outdated cant make a new install for my delta

Froggiexx, 2022-09-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

There is no updated delta config file for the new version of marlin, enabling delta under normal generic doesn't function and is outdated, using non normal delta and enabling delta in config wont properly work either for 2.1 marling 

### Bug Timeline

New from 2.1

### Expected behavior

It wont move the axis together at all or same speeds no working delta version or 2.1 

### Actual behavior

Y half works  works Y copes Z not all move it treats it like a cartisan printer not a delta 
even with delta enabled 

### Steps to Reproduce

have any custom delta run the marling in VS code using "new" delta generic config, and or normal config with delta enabled in the config all fail to work even know you can build the new one it wont move like  delta 

### Version of Marlin Firmware

2.1

### Printer model

Custom delta based on size of a He3d size not theirs my own 

### Electronics

SKR 1.3, 2209TMC's sensorless homing Bltouch and 24v power

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/9602335/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/9602336/Configuration_adv.txt)

thisiskeithb, 2022-09-20:
Can you clarify what you mean by "Delta all are outdated"?

Here are the latest `bugfix-2.1.x` Delta configs: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/delta

Here are the latest `2.1.1` Delta configs: https://github.com/MarlinFirmware/Configurations/tree/release-2.1.1/config/examples/delta

Neither are out of date when used with the correct version of Marlin:

`bugfix-2.1.x`: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x

`2.1.x`: https://github.com/MarlinFirmware/Marlin/tree/2.1.1

Froggiexx, 2022-09-20:
The delta confirms don’t work with marlin 2.1 

Cheers, 

Mike

thisiskeithb, 2022-09-20:
These configs build fine.

closed 2022-09-20 by thisiskeithb

ellensp, 2022-09-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Froggiexx, 2022-09-20:
Hi,  is there any wiki or article for BL touch on the delta?  I thought the
config for 2.1 was the right ones i didnt now they had to be the bug fix
ones so that was my bad im not a smart person or tech savvy this is hard
for me im trying to make it for my kid.   it moves correctly after the
right config, but doesnt home the Z or use the probe, sensorless homing is
enabled too.

cryptoAlgorithm, 2022-09-20:
@Froggiexx I recommend you ask in the Discord server or marlin forum, you'll get help much quicker there.