MarlinFirmware/Marlin #24781: [BUG] (bug summary)

cellularmitosis, 2022-09-20:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

It appears the TMC26XStepper library is now a 404:

```
Library Manager: Installing https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip
Error: Got the unrecognized status code '404' when downloaded https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip
```

This causes `1.1.9.1` to fail to build (using VS Code + PlatformIO).

### Bug Timeline

New

### Expected behavior

I expected the dependencies to not be 404.

### Actual behavior

One of the dependencies is a 404.

### Steps to Reproduce

- clone the repo
- `git checkout 1.1.9.1`
- click the checkmark button to build

### Version of Marlin Firmware

1.1.9.1

### Printer model

Ender 3

### Electronics

n/a

### Add-ons

n/a

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

cellularmitosis, 2022-09-20:
In fact, it appears the entire repo was removed.

Here is a wayback snapshot: https://web.archive.org/web/20201102024132/https://github.com/teemuatlut/TMC2208Stepper

Unfortunately, that particular zip file did not get archived (simply updating the URL would likely be the easiest solution).

cellularmitosis, 2022-09-20:
It appears this URL can be replaced with https://github.com/interactive-matter/TMC26XStepper/archive/refs/heads/master.zip

ellensp, 2022-09-20:
1.1.9.1 is a snapshot in time..  Ie it doesnt get updated

Marlin took a copy of this repository when it closed 
https://github.com/MarlinFirmware/TMC26XStepper

ie use https://github.com/MarlinFirmware/TMC26XStepper/archive/refs/tags/0.1.1.zip

TMC2208Stepper is not the same as TMC26XStepper

closed 2022-09-20 by ellensp

cellularmitosis, 2022-09-20:
Thank you for your quick reply.

I mistakenly though 1.1.9.1 was a branch.

Should I make a pull request against `bugfix-1.1.x` to get this fixed?

cellularmitosis, 2022-09-20:
Oh, this was already fixed in this commit: https://github.com/MarlinFirmware/Marlin/commit/02dcc0b80ed81f7263167cdb2fc68971fb3984e7

I guess that somehow didn't make it into the latest 1.1.x release

cellularmitosis, 2022-09-20:
Ah, it looks like `bugfix-1.1.x` isn't being used anymore, and `1.1.x` is the branch which is being updated.