MarlinFirmware/Marlin #24782: [BUG] Revisiting the 1.1.x sanguino_atmega1284p / ENDSTOP_INTERRUPTS_FEATURE issue (15540)

cellularmitosis, 2022-09-20:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

This is a continuation of https://github.com/MarlinFirmware/Marlin/issues/15540

At some point in 1.1.x, ATmega1284P-based boards (e.g. the Ender 3) starting hitting compiler errors, unless `ENDSTOP_INTERRUPTS_FEATURE` was disabled.

- https://github.com/MarlinFirmware/Marlin/issues/15540
- https://github.com/MarlinFirmware/Marlin/issues/15770
- https://www.danbp.org/p/en/node/146

This was thought to be addressed by https://github.com/MarlinFirmware/Marlin/pull/15771/files but pulling those changes into 1.1.x doesn't help for `sanguino_atmega1284p`.

The build error is the same reported in the above threads:

```
In file included from Marlin/endstops.cpp:35:0:
Marlin/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
Marlin/endstop_interrupts.h:138:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Users/cell/.platformio/packages/framework-arduino-avr/cores/arduino/Arduino.h:258:0,
                 from Marlin/HAL.h:36,
                 from Marlin/MarlinConfig.h:39,
                 from Marlin/Marlin.h:35,
                 from Marlin/endstops.cpp:27:
/Users/cell/.platformio/packages/framework-arduino-avr/variants/sanguino/pins_arduino.h:112:70: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)        ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                     ~^~~~~~~
Marlin/endstop_interrupts.h:138:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
```

However, the error about `reinterpret_cast from integer to pointer` is actually a misnomer, as this line of code should not have been reached, because these pins are interrupt capable.

That is, this line: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/endstop_interrupts.h#L107

```
  #if HAS_X_MIN
    #if digitalPinToInterrupt(X_MIN_PIN) != NOT_AN_INTERRUPT
      attachInterrupt(digitalPinToInterrupt(X_MIN_PIN), endstop_ISR, CHANGE);
```

should proceed along the `#if` case, because this pin is interrupt capable.

According to https://github.com/MarlinFirmware/Marlin/issues/15540#issuecomment-548178433 Arduino 1.8.8 was the last version under which this error did not exist.

So the real issue here is why did `digitalPinToInterrupt(X_MIN_PIN)` start returning `NOT_AN_INTERRUPT` after Arduino 1.8.8 for ATmega1284P (as defined by https://github.com/Lauszus/Sanguino/)?

Perhaps the modern version of avr-gcc requires something to be updated in the Sanguino board definition?

I do not expect the Marlin developers to spend time on this, but if there are other 1.1.x Ender 3 users out there, hopefully google will lead them here.

### Bug Timeline

old

### Expected behavior

I expected it to compile

### Actual behavior

It did not compile

### Steps to Reproduce

- git checkout 1.1.x
- copy in configs from https://github.com/MarlinFirmware/Configurations/tree/release-1.1.9/config/examples/Creality/Ender-3
- platformio run -e sanguino_atmega1284p

### Version of Marlin Firmware

1.1.x

### Printer model

Ender 3

### Electronics

Melzi 1.1.3

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-09-20:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

cellularmitosis, 2022-09-20:
I think I got it!

You just have to specify an older version of the toolchain in `platformio.ini`

```
[env:sanguino_atmega1284p]
platform      = atmelavr
platform_packages = toolchain-atmelavr@1.50400.190710
```

ellensp, 2022-09-20:
its some sort of issue with new compiler..

if you

```
#define MY_ATTACH_INTERRUP(p) ((p) == 2 ? 0 : ((p) == 3 ? 1 : ((p) >= 18 && (p) <= 21 ? 23 - (p) : NOT_AN_INTERRUPT)))

  #if HAS_X_MIN
    #if MY_ATTACH_INTERRUP(X_MIN_PIN) != NOT_AN_INTERRUPT

```
which should be the same as 

```
#define digitalPinToInterrupt(p) ((p) == 2 ? 0 : ((p) == 3 ? 1 : ((p) >= 18 && (p) <= 21 ? 23 - (p) : NOT_AN_INTERRUPT)))

  #if HAS_X_MIN
    #if digitalPinToInterrupt(X_MIN_PIN)) != NOT_AN_INTERRUPT

```

The former works, but the latter does not.

I suspect some sort of name conflict with digitalPinToInterrupt

cellularmitosis, 2022-09-20:
> The former works, but the latter does not.

Very interesting!  Thanks for taking a look.

For the time being I'll submit a PR to use `toolchain-atmelavr@1.50400.190710`.

closed 2022-09-20 by cellularmitosis

ellensp, 2022-09-20:
You should be using marlin 2

It compiles without issues on marlin 2.1.x bugfix

cellularmitosis, 2022-09-20:
Thanks, I'll give that a shot as well.