MarlinFirmware/Marlin #24800: [BUG] "LCD_BED_LEVELING is not supported by mks_robin_tft35

mr-september, 2022-09-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With the latest bugfix 2.1.x code on github, I'm getting this error on compilation.

"#error "LCD_BED_LEVELING is not supported by the selected LCD controller."

With my LCD controller being mks_robin_tft35. This used to work in 2.0.x.

### Expected behavior

Successful compilation

### Actual behavior

Compilation fails.

### Steps to Reproduce

#define LCD_BED_LEVELING

#define mks_robin_tft35

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Sapphire Plus

### Electronics

Stock

### Bed Leveling

MBL Manual Bed Leveling

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Also, the MKS Nano V1.2 mainboard used to be mks_nano_35 in platformio.ini, but it's now showing up as mks_robin_nanov1v2, mks_robin_nanov1v2_maple, and mks_robin_nanov1v2_usbmod. Is this right?
[Configuration & adv.zip](https://github.com/MarlinFirmware/Marlin/files/9640433/Configuration.adv.zip)

ellensp, 2022-09-25:
Not being supported is not a bug. 

you have 3 user interface choices
//#define TFT_CLASSIC_UI
//#define TFT_COLOR_UI
//#define TFT_LVGL_UI

only the top two support LCD_BED_LEVELING

thisiskeithb, 2022-09-25:
> Also, the MKS Nano V1.2 mainboard used to be mks_nano_35 in platformio.ini, but it's now showing up as mks_robin_nanov1v2, mks_robin_nanov1v2_maple, and mks_robin_nanov1v2_usbmod. Is this right?

Yes. This was changed in https://github.com/MarlinFirmware/Marlin/pull/24619

closed 2022-09-25 by thisiskeithb

mr-september, 2022-09-25:
Thanks for your replies. Perhaps the error message could be clearer: the incompatibility is from the UI, rather than the LCD controller.