MarlinFirmware/Marlin #24801: [FR] Known GPL violations list on webpage

uski, 2022-09-25:
### Describe the feature you want

Hi,

The Marlin firmware is released under the GPL.

I know of at least two manufacturers violating the Marlin license by not distributing the source code of their firmware :

**1) Creality**
The Ender 2 Pro is distributed with a motherboard using a microcontroller called HC32F460, which is not supported by the Marlin tree. As a result, they have ported Marlin to this microcontroller, but are not releasing the source code. There is a discussion in the Marlin github [here](https://github.com/MarlinFirmware/Marlin/issues/24338), and also Creality was asked to provide the source code without reply (I asked them by e-mail, they refused, and also it has been asked on their Github page, and they did not reply).

**2) Anycubic**
There are reports in [this Reddit thread](https://www.reddit.com/r/anycubic/comments/uz55jx/comment/ibupxfy/?utm_source=reddit&utm_medium=web2x&context=3) of some of their boards running Marlin, and they continuously claim they will release the firmware in the future without doing so.

At this point, these manufacturers are disrespecting the Marlin community, developpers/right holders, and their customers. They are preventing users from recompiling the firmware, which is essential to enable/disable features and make full use of their printers. I own an Ender 2 Pro and not having the firmware sourcecode is a problem to me, and it is also a problem for many other users.

I recommend that : 

1) We publish a list of known GPL violations on the Marlin website, pretty prominently. Maybe the bad press will make these manufacturers reconsider. This is very low effort, and even if it does not achieve the desired result of getting these manufacturers to comply with the GPL, it may discourage other manufacturers of pursuing the same violation.

2) We notify the manufacturers of their appearance on that list, and give them a chance to publish the firmware immediately. If they publish it within a week, we can remove them from the list. This gives them a chance to correct their behavior.

3) If they do not publish it within a week, we contact places such as the Software Freedom Conservancy. My understanding is that they have legal resources to go after manufacturers violatin the GPL. I know that Creality has at least some presence in the US, and is also being distributed by US companies such as Microcenter, so there is definitely something to go after on US territory. Even in China, GPL violations are successfully punished.

Does anyone has any other idea ? It seems that all of this is pretty doable, and if we don't defend Marlin and the GPL, noone is going to do it, and this situation will become more and more common.

thisiskeithb, 2022-09-25:
> Does anyone has any other idea ?

It would be great if we can get the usual 3D printing influencers to skip promoting products from known violators / hold off on posting related content until companies are in compliance.

uski, 2022-09-25:
I agree, and I think the first step is for us to publish known violations. We can then send the URL to the influencers, so that they are aware of the issue, and can continuously check that list before posting new content from a manufacturer.

I know some of them have already complained about GPL violations, so at least a few are already in agreement with that cause. We just need to compile a list of known offenders somewhere accessible. We can then contact the influencers

Roxy-3D, 2022-09-26:
I think I am supportive of these ideas.   But I would like to have a full discussion with the whole Dev-Team participating in it.
I like these ideas.   But I don't think it makes sense to do it unless we are serious and really are going to maintain the violator's list and making sure there is a penalty for bad behavior.

houseofbugs, 2022-09-26:
I'm tired of it as well. I setup a page and a reporting form for people to use. I'll maintain the list when I get time. https://gplviolations.th3dstudio.com/

gudvinr, 2022-09-26:
All that has been said so far is that violators should be punished but important thing here isn't to punish but to get compliance.  
And if manufacturers actually comply it is good to post some history of interaction and actually make them good guys if they have a good track record after they weren't.

Or even better, in addition to the list of bad guys there could be a list of good guys - vendors who go above and beyond. Who send pull requests upstream and/or regularly update their repos, sponsor marlin development, etc. That way vendors basically get positive exposure by doing what also benefits marlin in general.

EvilGremlin, 2022-09-26:
You're clearly new to this topic... with most violators the only way to get compliance is to punish them first - and some light public shaming doesn't really bother them, it gotta be something really big and painful, like US import ban.
Good guys are well known - and it's pretty much just board makers (MKS, BTT, etc).
Some others i know of:
Prusa (duh!) but they're not really relvant to vanilla marlin anymore in compliance sense.
Sovol publish everything (but i wish they didn't, it's prime facepalm material!), Geeetech used to but stopped (i don't know if they moved to their own firmware or just stopped caring)
Rrandom nonames from Ali i can't be bothered to remember are surprisingly responsive and often send code at first ask. At least i've seen decent amount of sch reports.
Anyway, 'good guys' list is good idea.

gudvinr, 2022-09-26:
I am not saying that there shouldn't be any punishment for violators. I am saying that punishment itself isn't the goal but rather a way to get compliance. When vendors are ignorant and not the first time violators, actions should be taken.  

But if there is nothing positive to counter that it might give marlin a bad look in a sense that it only shows toxicity towards vendors. So if vendors can get either nothing or bad exposure, they might start thinking about developing more proprietary firmware. Especially big ones, including creality or anycubic who might have resources for that.

Although I agree that "good guys" list requires thorough thinking to develop criteria that are actually meaningful and abuse-proof yet not that hard to meet.  
E.g. "doing X pull requests a year" will lead to useless PRs to deal with but something like "upstreaming high-quality code for new printers" is another thing. And said violators can be banned from this list until they prove their reliability which might take a long time.

houseofbugs, 2022-09-27:
> I am not saying that there shouldn't be any punishment for violators. I am saying that punishment itself isn't the goal but rather a way to get compliance. When vendors are ignorant and not the [...]

If you want to compile a list of vendors and models that comply I'll gladly add a section to the page to show compliance and link to their source. I dont have time for that but if someone else will compile the information I'll gladly add it and host it on the site.

classicrocker883, 2023-04-01:
anyone able to get the Hc32f460 to work like any other chip in platformio https://github.com/shadow578/Marlin-H32