MarlinFirmware/Marlin #24802: [BUG] compile without sd card support

Gueulfor, 2022-09-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I detect an error while i compile your latest bug-fixed firmware.I'm on homemade printer without SD card.At line 1852 of temperature.cpp methode "task" if you don't use "SDSUPPORT" compilation failed because of "sd_abort_by_M524" member of object "EmergencyParser" are undeclared.

### Bug Timeline

new bug

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

reprap

### Electronics

ramps 1.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I solve the problem with #ifdef on lines using "sd_abort_by_M524" on file "temperature.cpp"

thisiskeithb, 2022-09-25:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24774

thisiskeithb, 2022-09-25:
This will be fixed once https://github.com/MarlinFirmware/Marlin/pull/24775 is merged.

closed 2022-09-25 by thisiskeithb