MarlinFirmware/Marlin #24805: [BUG] Firmware configuration file does not account for HR4988 driver

robotom, 2022-09-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My SD Card slot is marked "C" which means the driver for the motors is HR4988. However, in the Ender 3 Pro configuration file, this is not a specified option. It lists A4988 but not the HR4988. I don't want to risk bricking the printer. Board version is 4.2.2.

Any idea what I should be writing in the configuration file? Thanks. 

### Version of Marlin Firmware

2.1

### Printer model

Ender 3 Pro

### Electronics

4.2.2

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-09-25:
Use `A4988`

thisiskeithb, 2022-09-25:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23626

closed 2022-09-25 by thisiskeithb

robotom, 2022-09-25:
> Use `A4988`

Thanks! I presume the timing is the same, just noisier on my older steppers?

thisiskeithb, 2022-09-25:
From https://github.com/MarlinFirmware/Marlin/issues/23626#issuecomment-1162018869:

> The thing is that as far as Marlin is concerned the only applicable settings for A4988 and HR4988 are the pulse timings, and those are identical. So there's no such thing a "more sane" option to use for an HR4988 on the Creality boards. The micro stepping is not configurable on those boards so there's nothing Marlin can do to change it.

robotom, 2022-09-25:
Compiles and works fine! (After some tweaks). The version stipulated in the code is something like 02010000 so if you go to compile with the latest source files it throws a few dozen errors. Marlin is 2.1.1 so I had to change that to 02010100. Might want to add that in the documentation somewhere. I presume the devs have too many printers and firmware versions to keep up with. Thanks for the help.

thisiskeithb, 2022-09-25:
Marlin and Configuration versions need to match. 

Examples:

Marlin `2.1.1` Release: https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.1

Used with Configurations Release `2.1.1`: https://github.com/MarlinFirmware/Configurations/tree/release-2.1.1

<hr>

Marlin `bugfix-2.1.x`: https://github.com/MarlinFirmware/Marlin

Used with Configurations `bugfix-2.1.x`: 
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x

etc…