MarlinFirmware/Marlin #24808: [FR] Z_STEPPER_AUTO_ALIGN add comment for common problem

drakepro, 2022-09-26:
### Is your feature request related to a problem? Please describe.

When enabling Z_STEPPER_AUTO_ALIGN it is possible for the z steppers to not be able to adjust the x axis into alignment because the hardware is out of square and all the screws are too tight.  It would be helpful to add a comment to the configuration_adv.h to indicate that the user may want to loosen the screws that dictate the squareness of their X axis in relation to their Z.

I have found a few examples of people online frustrated with this issue, as I was.  Loosening the screws that attach the X axis to the Z axis, then running G34, then tighten the screws back up completely solved my issue.

### Describe the feature you want

Add a comment above Z_STEPPER_AUTO_ALIGN in Configuration_adv.h to tell user they may need to loosen their X -> Z axis screws so that the Z steppers may physically be able to level the Z axis.

ellensp, 2022-09-26:
Mechanical issues are beyond the purview of Marlin firmware.

closed 2022-09-26 by ellensp

EvilGremlin, 2022-09-26:
i'll add that Z nuts must be decoupled from gantry in XY plane anyway (antiwobble nut)