MarlinFirmware/Marlin #24812: [BUG] Auto bed leveling not works in Biqu B1 SE

vovodroid, 2022-09-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bed leveling stops at second point without touching. After  a while system goes to the main screen.

I have built firmware for Biqu B1 SE basing on config of B1 SE Plus https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/B1%20SE%20Plus with changed bed size and and enabled ALLOW_LOW_EJERK (though I don't think it matters).

### Expected behavior

Nozzle touches all needed points and plate mesh is stored.

### Actual behavior

After clicking on Leveling in the menu head goes to left bottom corner, touches the plate, next move to right about 50-60 mm and stops. After a while screen returns to the main window with message "Printer is ready" appearing.

### Steps to Reproduce

Go to Moves menu, choose Leveling.

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Biqu B1 SE

### Electronics

Stock BTT SKR 2 Rev B

### Bed Leveling

ABL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9668175/Marlin.zip)

thisiskeithb, 2022-09-28:
In https://github.com/MarlinFirmware/Configurations/pull/814#issuecomment-1260900935, you said that you modded the printer to convert it to a direct drive. Did you reuse the existing strain gauge hotend? Since the SE & SE Plus are bowden-based and use the nozzle as a probe (`NOZZLE_AS_PROBE`), the firmware in the strain gauge / hotend will not work for homing/probing.

Photos of the machine & whatever other mods you have done would help. Also, whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

vovodroid, 2022-09-29:
> Did you reuse the existing strain gauge hotend?

Sure, I just printed mount for stock extruder and put it above stock hotend without any other modifications:
[image]

So I tried the steps your suggested. It fails with error ```FAST Probe fail! - Triggered early.```, first time at 12th point, second at 2nd, log attached. Stock firmware works:

```
M420 V0
Bilinear Leveling Grid:
      0      1      2      3      4      5      6      7      8
 0 +0.140 +0.111 +0.037 +0.008 -0.068 -0.077 +0.001 +0.066 +0.087
 1 +0.148 +0.084 +0.041 +0.005 -0.063 -0.074 +0.001 +0.034 +0.090
 2 +0.160 +0.101 +0.049 -0.000 -0.025 +0.005 +0.032 +0.114 +0.121
 3 +0.170 +0.139 +0.104 +0.055 +0.034 +0.003 +0.084 +0.138 +0.144
 4 +0.181 +0.131 +0.122 +0.100 +0.089 +0.083 +0.136 +0.170 +0.225
 5 +0.189 +0.125 +0.146 +0.096 +0.114 +0.104 +0.190 +0.232 +0.273
 6 +0.202 +0.202 +0.184 +0.151 +0.173 +0.134 +0.222 +0.283 +0.333
 7 +0.211 +0.176 +0.209 +0.197 +0.237 +0.190 +0.308 +0.364 +0.414
 8 +0.222 +0.178 +0.215 +0.171 +0.241 +0.158 +0.304 +0.340 +0.414
```

[putty.log](https://github.com/MarlinFirmware/Marlin/files/9671391/putty.log)

thisiskeithb, 2022-09-29:
The firmware on the strain gauge in the hotend is not calibrated for anything other than the stock (lighter) bowden hotend, so it will not work.

Since the firmware on the strain gauge is not open source and this particular hardware revision only exists on the B1 SE beta units, you will need to disable the strain gauge & use a real probe.

closed 2022-09-29 by thisiskeithb

vovodroid, 2022-10-14:
It sometimes also fails even without extruder mounted, but this time at slow speed ```SLOW Probe fail! - Triggered early.```, so it's not mater of weight (or probably more exactly matter of wobbling because lack of rigidity of mount).

```
>  X215.0000 Y175.0000 Z1.3675
echo:busy: processing
<<< do_blocking_move_to  X215.0000 Y175.0000 Z1.3675
  current_position= X215.0000 Y175.0000 Z1.3675 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X215.0000 Y175.0000 Z1.3675
>>> Probe::probe_down_to_z  X215.0000 Y175.0000 Z1.3675
>>> do_blocking_move_to  X215.0000 Y175.0000 Z1.3675
>  X215.0000 Y175.0000 Z-2.0500
echo:busy: processing
<<< do_blocking_move_to  X215.0000 Y175.0000 Z-2.0500
  current_position= X215.0000 Y175.0000 Z0.5325 : sync_plan_position
<<< Probe::probe_down_to_z  X215.0000 Y175.0000 Z0.5325
SLOW Probe fail! - Triggered early.
<<< Probe::run_z_probe  X215.0000 Y175.0000 Z0.5325
  current_position= X215.0000 Y175.0000 Z0.5325 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X215.0000 Y175.0000 Z0.5325
>  X215.0000 Y175.0000 Z0.5325
<<< do_blocking_move_to  X215.0000 Y175.0000 Z0.5325
Error:Probing Failed
measured_z: 0.0000

```

thisiskeithb, 2022-10-14:
Again...

> Since the firmware on the strain gauge is not open source and this particular hardware revision only exists on the B1 SE beta units, you will need to disable the strain gauge & use a real probe.

We cannot support this *beta hardware* and it was never meant for general/public use.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).