MarlinFirmware/Marlin #24816: [BUG] M218 X# Y# causes toolchanger to fail

aamott, 2022-09-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On a 3D printer with `SWITCHING_TOOLHEAD` enabled, if a hotend offset is set using `M218 T1 X# Y#`, it is not applied when loading (`T1`), but _is_ still applied when unloading the tool (`T-1` or `T0`), resulting in the tool missing its mark and crashing into mount. The X and Y offsets should be ignored when loading _and_ unloading the tool. 

### Bug Timeline

Unknown

### Expected behavior

When `M218` is used to set an X or Y offset on Tool 1 (or T2, T3...) and `T1` is run, it should park the current tool, pick up the new tool, apply offset, and return the tool.

Upon running `T0`, it should do the same in reverse (offset for T0 being nothing).

### Actual behavior

The first tool change is successful (`T1`). It loads the tool then applies the offset. On return (`T0`) the offset remains until after the tool change is finished. It tries to return the tool, but 5mm to the right and 0.7mm too far back. In testing by removing the tool, it also tries to pick up the new tool 5mm right and 0.7mm too far back. Each will result in a crash. 

### Steps to Reproduce

1. Set a tool offset: `M218 T1 X-5`
2. Home: `G28`
3. Load tool 1: `T1`
   Tool should load correctly
5. Load tool 0: `T0`
     Tool tries to unload and load 5mm to the right

### Version of Marlin Firmware

2.1.x

### Printer model

Custom CoreXY

### Electronics

Printrboard Rev E

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9671101/configurations.zip)

For a visual, here's the toolchanger: [Compact CoreXY Toolchanger](https://grabcad.com/library/compact-corexy-toolchanger-1)

DerAndere1, 2022-10-01:
try the following changes: in file tool_change.cpp lines 521-716

In all lines starting with `current_position.x`,  insert the following before the semicolon that terminates the line:
`+ hotend_offset[active_extruder].x`

e.g. change 
https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/src/module/tool_change.cpp#L521

to this:
```current_position.x = placexpos + hotend_offset[active_extruder].x;```

Similarly,  in all lines starting with `current_position.y`,  insert the following before the semicolon that terminates the line:
`+ hotend_offset[active_extruder].y`

aamott, 2022-10-03:
That worked great! Every other toolchanger has offset compensation, but each one is a little bit different. I wonder if this is worth standardizing between each section.
`offsetcompensation` is defined for the `MAGNETIC_PARKING_EXTRUDER` [here](https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/src/module/tool_change.cpp#L173). `PARKING_EXTRUDER` uses `x_offset` [here](https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/src/module/tool_change.cpp#L297). `ELECTROMAGNETIC_SWITCHING_TOOLHEAD` [has it](https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/src/module/tool_change.cpp#L736) called `&hoffs`, including a y offset. Finally, `SWITCHING_NOZZLE_TWO_SERVOS` [has it](https://github.com/MarlinFirmware/Marlin/blob/e1df912dc7f05858c727d4e0994082b78ae1998f/Marlin/src/module/tool_change.cpp#L1214).

thisiskeithb, 2022-10-03:
Closing since you've opened a PR:

- https://github.com/MarlinFirmware/Marlin/pull/24831

closed 2022-10-03 by thisiskeithb