MarlinFirmware/Marlin #24817: Klicky,  G28, unable to control the movement for Z before probing.

alfskaar, 2022-09-29:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Klicky,  G28, unable to control the movement for Z before probing.

I'm giving up on this configuration that use the Klicky "MAG_MOUNTED_PROBE".
I have not trided the latest bugfix build, however unless the way Marlin do G28 have change I don't see it will change anything.

The biggest problem is when it moves to the front left X0, Y0, Z0 this is after it have picked up the Klicky probe it then moves down the Z and if the gantry on the Voron 2.4 R2 is to close to the bed it will hit the bed and the Klicky probe will fall off, the only way It seems possible to control this is to make sure the gantry is high enough above the bed befor the G28 start.

I have include the debug log since I may do more testing I have named all files 01_*.*

One other thing that is really strange is that the end stop is all connected to the top row on the PCB
but still they are defined as USE_XMAX_PLUG, USE_YMAX_PLUG, USE_ZMIN_PLUG it seems to me looking in the "pins_BTT_OCTOPUS_V1_common.h" it looks like the same address is used for both.
If I move like the USE_XMAX_PLUG down to the bottom row on the PCB it doesn't work.

Only way I can understand this is that the BTT OCTOPUS PRO only have one diag pin (for use on min X or max X) and not two like my "Mini-Rambo 1.3a" have.

### Bug Timeline

new

### Expected behavior

To probe and not move down Z

### Actual behavior

it moves the Z down before it actually move to the prob point.

### Steps to Reproduce

1. Voron 2.4 R2
2. add my config files and compiled it.
3. G28

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Voron 2.4. R2

### Electronics

BTT OCTOPUS PRO

### Add-ons

Klicky

### Bed Leveling

ABL 3-point

### Your Slicer

Slic3r

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-09-29:
> I have not trided the latest bugfix build

Please do since we have no idea what version of Marlin you are running.

Speaking of:

> Don't forget to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You forgot to do this.

alfskaar, 2022-09-29:
[01_debug.zip](https://github.com/MarlinFirmware/Marlin/files/9672204/01_debug.zip)

alfskaar, 2022-09-29:
Tested the "Marlin-bugfix-2.1.x.zip" that I just download no change I do use the same config files and change the "CONFIGURATION_H_VERSION 02010200" on my files so the config is the same. Will check if there is something new in the. new config files that I have missed.

alfskaar, 2022-09-29:
MINIRAMBO have two pins BTT PRO have only one for every stepper. Can this be the problem ? since I home to MAX ?

pins_MINIRAMBO.h
//
// Limit Switches
//
#define X_MIN_PIN                             12
#define X_MAX_PIN                             30
#define Y_MIN_PIN                             11
#define Y_MAX_PIN                             24
#define Z_MIN_PIN                             10
#define Z_MAX_PIN                             23

pins_BTT_OCTOPUS_V1_common.h
//
// Trinamic Stallguard pins
//
#define X_DIAG_PIN                          PG6   // X-STOP
#define Y_DIAG_PIN                          PG9   // Y-STOP
#define Z_DIAG_PIN                          PG10  // Z-STOP
#define Z2_DIAG_PIN                         PG11  // Z2-STOP
#define E0_DIAG_PIN                         PG12  // E0DET
#define E1_DIAG_PIN                         PG13  // E1DET
#define E2_DIAG_PIN                         PG14  // E2DET
#define E3_DIAG_PIN                         PG15  // E3DET

BIGTREETECH Octopus Pro - PIN.pdf
stop0	stop1	stop2	stop3
PG6	PG9	PG10	PG11
Xmin	Ymin	Zmin
(min or max)
Xmax	Ymax	Zmax	

stop4	stop5	stop6	stop7
PG12	PG13	PG14	PG15

alfskaar, 2022-10-12:
/**
 * Magnetically Mounted Probe
 * For probes such as Euclid, Klicky, Klackender, etc.
 */
#define MAG_MOUNTED_PROBE
#if ENABLED(MAG_MOUNTED_PROBE)
  #define PROBE_DEPLOY_FEEDRATE (133*60)  // (mm/min) Probe deploy speed
  #define PROBE_STOW_FEEDRATE   (133*60)  // (mm/min) Probe stow speed
                                                        // X, Y, Z, all moves need to complete
  #define MAG_MOUNTED_DEPLOY_1 { PROBE_DEPLOY_FEEDRATE, { 52, 180,  20 } }  // Move to side Dock & Attach probe
  #define MAG_MOUNTED_DEPLOY_2 { PROBE_DEPLOY_FEEDRATE, { 52, 202,  20 } }  // Move probe off dock
  #define MAG_MOUNTED_DEPLOY_3 { PROBE_DEPLOY_FEEDRATE, {  0,   0,  20 } }  // Next move
  #define MAG_MOUNTED_DEPLOY_4 { PROBE_DEPLOY_FEEDRATE, {  0,   0,  20 } }  // Next move
  #define MAG_MOUNTED_DEPLOY_5 { PROBE_DEPLOY_FEEDRATE, {  0,   0,  20 } }  // Next move
  #define MAG_MOUNTED_STOW_1   { PROBE_STOW_FEEDRATE,   { 52, 160,  20 } }  // Move to dock
  #define MAG_MOUNTED_STOW_2   { PROBE_STOW_FEEDRATE,   { 52, 200,  20 } }  // Place probe beside remover
  #define MAG_MOUNTED_STOW_3   { PROBE_STOW_FEEDRATE,   { 56, 200,  20 } }  // Side move to remove probe
  #define MAG_MOUNTED_STOW_4   { PROBE_STOW_FEEDRATE,   { 56, 180,  20 } }  // Side move to remove probe
  #define MAG_MOUNTED_STOW_5   { PROBE_STOW_FEEDRATE,   {   0,   0, 20 } }  // Next move if needed
#endif

It works, you can close it. But where are the Z_OFFSET menu how ? @thisiskeithb

alfskaar, 2022-10-13:
.

closed 2022-10-13 by alfskaar