MarlinFirmware/Marlin #24820: [BUG] Bilinear Bed Levelling = incorrect Z offsets for first 10mm of Z travel, squashed prints

westhedge, 2022-09-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I noticed on a new build that my prints suddenly were squashed by about 4mm once I had enabled bilinear bed levelling. As it turns out, the 2.1.1 firmware ABL corrections are giving non-linear z offsets for the first 10mm of z, then the offsets are constant after. For example, if I enable ABL and set absolute positioning and go through z 0-30 (first column below) The actual Z position (as measured from an arbitrary reference point on my printer) is in column 2. Column 3 is just the difference (i.e. the offset from where I tell it to go and where it actually goes. This test was run with the X/Y in the same position. 

So it appears ABL is either broken or there is some non-obvious configuration setting that needs to be applied. Please let me know if you need any information. I am providing the configuration files.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/anels/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/anels/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Specified | Actual | Z Offset
-- | -- | --
0 | -10.65 | -10.65
2 | -9.3 | -11.3
4 | -8.1 | -12.1
5 | -7.39 | -12.39
6 | -6.9 | -12.9
8 | -5.5 | -13.5
10 | -4.05 | -14.05
12 | -2.35 | -14.35
15 | 0.7 | -14.3
20 | 5.8 | -14.2
25 | 10.8 | -14.2
30 | 15.9 | -14.1

</body>

[image]

</html>
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9678496/marlin.zip)

### Bug Timeline

New issue - just built this firmware

### Expected behavior

The Z-offset from absolute positioning G0 Z{position} should be constant from Z=0 to Z=infinity

### Actual behavior

Z-offset is not constant and prints get squashed in the first 10mm

### Steps to Reproduce

G28
M420 S1
G0 Z0
;measure actual Z-height against arbitrary reference point with dial indicator and calculate "offset"
G0 Z2
;measure actual Z-height against arbitrary reference point with dial indicator and calculate "offset"
G0 Z4
;and repeat measurement, etc...
; ...
G0 Z20
;measure, repeat

### Version of Marlin Firmware

2.1.1

### Printer model

Stratasys

### Electronics

Bigtreetech Octopus

### Add-ons

BL Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

westhedge, 2022-09-29:
This appears to be an issue with ENABLE_LEVELING_FADE_HEIGHT. If I disable this in firmware, offsets are applied correctly. I'm reasonably certain this is still a bug.

ellensp, 2022-09-30:
This is how it is meant to work

```
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT
  #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
    #define DEFAULT_LEVELING_FADE_HEIGHT 10.0 // (mm) Default fade height.
  #endif

```

closed 2022-09-30 by ellensp

westhedge, 2022-09-30:
Why would anyone ever want a feature that produces an inaccurate print? This makes zero sense. In my case, the standard 20x20x20 calibration cube ended up being 20x20x16. Not even close to accurate dimensions. What does make sense to me is that as the gantry goes up in Z, the presumably minute levelling correction would become gradually faded out until the fade height is reached, thus traversing the bed in X/Y at this height would require no Z movement. Effectively this would squash or raise "warped" sections of layers until it is gradually levelled out at the fade height.  That's not what actually happens, though.

westhedge, 2022-09-30:
And here are 3 posts where other people have experienced a similar issue. A picture is worth a thousand words: notice how the calibration cube is stretched or squished depending on ABL's Z fade height:
https://github.com/MarlinFirmware/Marlin/issues/7456
https://github.com/MarlinFirmware/Marlin/issues/9118
https://3dprinting.stackexchange.com/questions/16793/creality-ender-3-pro-z-travel-issue-with-th3d-firmware

senjabl, 2022-10-11:
@westhedge , do you possibly have Zmax endstop?
In my setup with heatbed going up/down, I have Zmax endstop. I spent hours fiddling with Fade height, getting incorrect height of my prints. 
Workaround is I had to tune Z_MAX_POS in configuration.h as accurate as possible (mine is now 216.3 mm). 
If you turn off Bed leveling and do G1 Z0 (or just move heatbed to Z=0), distance between heatbed and nozzle will be added/subtracted to Fade height. 
That means, if your Fade height is 10mm and on Z=0 you have eg 4mm between nozzle and heatbed, that what is suppose to be first 10 mm of print will be squashed to 6mm. On the other side if heatbet touches nozzle on eg. Z=4 mm print will be 4mm higher.

In Repetier firmware, there is option to measure Z MAX  with probe and store it to EEPROM