MarlinFirmware/Marlin #24822: [BUG] (bug summary)

acparker18, 2022-09-30:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

M500 save to EEPROM does not function unless executed twice.

This appears to be a reintroduction of this older, resolved bug:

#17627

Working now to test on newer 2.1.x nightly.

### Bug Timeline

New with 2.0.9.2

### Expected behavior

M500 saves current configuration to EEPROM after executing only once. Either from a command sent over serial, or from the printer interface/screen.

### Actual behavior

M500 shows successful save, but M503 reports old values. New values are not saved to EEPROM until M500 is executed a second time.

### Steps to Reproduce

1. Make changes, such as adjusting ESteps value
2. Execute M500 directly or via some method on the printer interface/screen
3. Execute M503 and notice that the new values did not save.
4. Make the changes again
5. Execute M500
6. Execute M500 again
7. Execute M503 and notice that the values saved correctly to EEPROM this time

### Version of Marlin Firmware

2.0.9.2

### Printer model

Anycubic Vyper

### Electronics

Stock

### Add-ons

N/A

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

acparker18, 2022-09-30:
[VyperCE_Config.zip](https://github.com/MarlinFirmware/Marlin/files/9688074/VyperCE_Config.zip)

thisiskeithb, 2022-11-30:
Support for `BOARD_AC_TRI_F1_V1` has not been added here your configs are from [Pmatsol58/VyperCE6.1](https://github.com/Pmatsol58/VyperCE6.1), so you'll need to file a bug report with them.

closed 2022-11-30 by thisiskeithb