MarlinFirmware/Marlin #24823: [BUG] (Automatic switching of the extruder motors does not work)

DoctorDeatch, 2022-10-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when using two extruders, only one works, which is initially activated in the control system.
the heating of the nozzles is going correctly, but only one motor works

this happens when you use both extruders, or the second extruder separately

### Bug Timeline

old

### Expected behavior

the operation of the extruder motor depends on the extruder to which the print is directed

### Actual behavior

this extruder motor is activated, which is initially selected in the system, program or TFT display

### Steps to Reproduce

Steps to Reproduce

### Version of Marlin Firmware

2.1.1

### Printer model

Flyingbear 3D Printer P905H

### Electronics

MKS Robin Nano V2

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[https://disk.yandex.ru/d/MzZkIhEWLT9wkA](config)

ellensp, 2022-10-01:
Local copy of files uploaded [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9689735/Marlin.zip)

ellensp, 2022-10-01:
where are the "Steps to Reproduce" ?

Have you done any basic testing?

You say "both extrudes" but there is no mixing setup, so be can only be T0 or T1   (ie E0 or E1)
And not IDEX  (DUAL_X_CARRIAGE) ...   so both does not seem to apply.

basic gcode test
T0
M104 S200  (or what ever temperature)
G1 E5     extrude some filament from E0
T1
M104 S200  (or what ever temperature)
G1 E5    extrude some filament from E1

DoctorDeatch, 2022-10-02:
the steps for reproduction are just to try to print from the second extruder
, I do not have a mixing nozzle, but two separate ones (as can be seen from the config)
when trying to use the second nozzle. regardless of whether the motor of the 1 extruder works together with the first one (two-color printing) or independently, it can only be switched manually in the rasbury host program.

ellensp, 2022-10-02:
the config says BOARD_MKS_ROBIN_NANO_V3
but you have said Electronics MKS Robin Nano V2

Which is correct?

DoctorDeatch, 2022-10-02:
corrected 
MKS Robin Nano V3