MarlinFirmware/Marlin #24830: [BUG] (Dual Z axis - No homing on second axis on second driver, but z commands move both z)

CandidQuality, 2022-10-03:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Likely in the wrong place, but at my wits end, would appreciate any help on what obvious error I'm creating for myself. 
I simply cannot fathom what part of the code I've missed to force both z axis to home in unison.  
 
[FIRMWARE.CUR.inverted.stops.to.NO.from.zip](https://github.com/MarlinFirmware/Marlin/files/9701493/FIRMWARE.CUR.inverted.stops.to.NO.from.zip)

### Expected behavior

E expect both z to home and trigger on end-stop

### Actual behavior

only the left axis moves on the home command when using pronterface,  

### Steps to Reproduce

1. compiled based on the attached files
2. load build onto sd card
3. successful flash of new file
4. testing results in second z running correcty on move command, but home command simply does not move second z motor.  

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anycubic  Mega

### Electronics

SKR3

### Add-ons

No addons

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I really do apologize if this is in the wrong forum, I'm completely lost.  just grasping for straws.   Feel free to delete and tell me what simple step I'm missing if this is not actually an error.  First time compiling Marlin.  Worked my way through a dozen other errors before getting a good compile, so this build has no errors on compile.   

Have also tried copying BTT recommended profile for 2.0.1 to the latest 2.1 bug-fix, and cannot count the amount of errors I've tried to re-create from the pre-configured file. so just trying to get anything to work here.

ellensp, 2022-10-03:
normally this line   `#define Z2_USE_ENDSTOP          _ZMAX_`
When the second Z endstop is plugged in elsewhere..

This ZMAX is the PWRDET port

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-10-03 by ellensp

CandidQuality, 2022-10-03:
Completely understand, thank you for pointing me to a non-facebook forum.  Trying on reprap forum now.  I appreciate your patience.  

 #define Z2_USE_ENDSTOP          _ZMAX_  // from here down it shows the code as being inactive no matter how many stepper drivers I try to select.  so the zmax endstop is not actually defining?