MarlinFirmware/Marlin #24833: [BUG] compile fails when TEMP_SENSOR_BED set to not used

mAd-DaWg, 2022-10-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Configured for a cnc machine. Compiling fails when TEMP_SENSOR_BED is set to 0. When its set to a dummy sensor it compiles.

### Bug Timeline

newish issue as far as i am aware

### Expected behavior

Expect the firmware to compile without issue when bed sensor is set to unused

### Actual behavior

firmware fails to compile.

`Compiling .pio/build/BIGTREE_SKR_2/src/src/sd/cardreader.cpp.o
Compiling .pio/build/BIGTREE_SKR_2/src/src/sd/usb_flashdrive/Sd2Card_FlashDrive.cpp.o
Marlin/src/module/temperature.cpp: In static member function 'static void Temperature::task()':
Marlin/src/module/temperature.cpp:1855:7: error: 'gcode' was not declared in this scope
 1855 |       gcode.process_subcommands_now(F("M524"));
      |       ^~~~~
*** [.pio/build/BIGTREE_SKR_2/src/src/module/temperature.cpp.o] Error 1
============================================================================ [FAILED] Took 34.39 seconds ============================================================================

Environment    Status    Duration
-------------  --------  ------------
BIGTREE_SKR_2  FAILED    00:00:34.387
======================================================================= 1 failed, 0 succeeded in 00:00:34.387 ==`

### Steps to Reproduce

1. Set temp sensors to 0
2. Attempt to compile

### Version of Marlin Firmware

Marlin-bugfix-2.1.x (18 Sep 2022)

### Printer model

Custom built cnc

### Electronics

SKR-2 rev b

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9718898/Configuration.zip)

EvilGremlin, 2022-10-05:
show full log with first error, you're just showing last eroor which is usually random meaningless garbage

EvilGremlin, 2022-10-05:
there are many things wrong with this config, but temp_sensor is not one of them

EvilGremlin, 2022-10-05:
oooh, i got it - problem with include conditions
temperature.cpp - change line 66 to `#if ANY(HAS_TEMP_SENSOR, LASER_FEATURE, SPINDLE_FEATURE)`

EvilGremlin, 2022-10-05:
fixed in https://github.com/MarlinFirmware/Marlin/pull/24834

closed 2022-10-05 by thisiskeithb