MarlinFirmware/Marlin #24835: [BUG] Bilinear autoleveling last row is far from bed Y edge

vovodroid, 2022-10-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Last row of probes is far away (60 mm) from max Y bed edge despite ```#define PROBING_MARGIN 20``` 

[image]

### Expected behavior

Last row has the same distance from bed edge, as the first one.

### Actual behavior

Last row has 60 mm from bed edge, while the first has 20 mm.

### Steps to Reproduce

Not sure for all cases, but in my they are:
1) Define and build
```
#define AUTO_BED_LEVELING_BILINEAR
#define PROBING_MARGIN 20 

#define X_BED_SIZE 235
#define Y_BED_SIZE 235

#define X_MIN_POS -7
#define Y_MIN_POS -3
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE + 5
#define Z_MAX_POS 270
```

2) Issue G29

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Biqu B1 SE

### Electronics

Stock board

### Bed Leveling

ABL Bilinear mesh

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Bed size is 235, bed leveling bilinear, 25 points, 20 mm margin.

So X/Y steps should be (235-20*2)/4 = 48.75 
In reality Y step is about 38.75 mm, thus last Y pobe is at 136, and not 215.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9722277/Marlin.zip)
[putty.log](https://github.com/MarlinFirmware/Marlin/files/9722280/putty.log)

thisiskeithb, 2022-10-06:
As I've said in your other issues, this printer is beta hardware from Biqu and is not available for public / retail. Please contact Biqu directly if you were/are a beta tester to resolve issues with this printer.

Having said that, this is a basic i3, so something about your travel limits are likely off and this is not a bug.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2022-10-06 by thisiskeithb

vovodroid, 2022-10-06:
> this printer is beta hardware from Biqu and is not available for public / retail.

Hi, I don't really understand what is link to Biqu hardware? Grid steps/positions are calculated by Marlin on the base of bed size and margins, aren't they?

And I have my own Marlin build, not Biqu's. Besides the fact that SKR 2 board is not unique to B1 SE.

vovodroid, 2022-10-15:
> this printer is beta hardware from Biqu

It's not hardware: https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/BIQU/B1%20SE%20Plus/Configuration_adv.h#L2144

```
#if PROBE_SELECTED && !IS_KINEMATIC
  //#define PROBING_MARGIN_LEFT PROBING_MARGIN
  //#define PROBING_MARGIN_RIGHT PROBING_MARGIN
  //#define PROBING_MARGIN_FRONT PROBING_MARGIN
  #define PROBING_MARGIN_BACK 60
#endif
```

thisiskeithb, 2022-10-15:
@vovodroid: You've been told several times to stop using this issue queue as tech support. Use the links in my previous comments for config setup & troubleshooting because **this is not a Marlin bug**.

vovodroid, 2022-10-15:
I just wanted to say, that not all issues with B1 SE is related to Biqu hardware, and it's not always easy to distinguish between code bug and just configuration effect.

Anyway I would like to thank you for pointing me to the right direction!