MarlinFirmware/Marlin #24840: [BUG] (bug summary)

unitrix0, 2022-10-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have installed a "HeroMe Gen5" on my Ender 5 Plus which requires me to set the home offset on the Y axis to -9. Now, recently I discovered that the G30 command doesn't seem take this offset into account.

### Bug Timeline

Discovered in 2.0.9.3

### Expected behavior

When a G30 command is executed the probe should measure a the location where the nozzle would be. For example `G30 X180 Y180` should measure the point at which the nozzle would be when `G0 X180 Y180` is executed. 

### Actual behavior

The probe measures at a different location, in my case 9mm off on the Y axis.

### Steps to Reproduce

1. Configure a Home Offset on X or Y axis
2. Execute a G30 command for any location

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 5 Plus

### Electronics

Bigtreetech SKR MINI E3 V3.0

### Add-ons

Hero Me Gen5

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9735013/Configurations.zip)

ellensp, 2022-10-07:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24590

closed 2022-10-07 by ellensp

mriscoc, 2022-11-18:
Home offset has effect on work area not on probing area, that's why I recommend to use XYZ Min/Max position settings instead of Home offsets.
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-set-the-home-offsets