MarlinFirmware/Marlin #24841: [BUG] PID tuning behavior / PID settings ignored?

parallyze, 2022-10-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi all,

for the past days I've been trying something very simple - setting correct PID values on this printer:

i3 Style, SKR 2 Rev. B, TMC2208/UART, Dual Z/G34, 40W Heater, Single Extruder

The machine was running Marlin 2.1.1 release. So far it is working great, I just noticed a minor
thing: Temperature was oscillating while printing and when heating up it did look like wrong PID
parameters.

Running PID autotune and copy/pasting the values into the config files or using M301/M500 does always
result in the same strange behavior I somehow don't understand right now.

After power cycling heating the extruder to 200°C will ALWAYS slow down around ~195°C, peak at
196/197 and then overshoot massively to 206/207. 

The moment I set PID values using M301 it will always overshoot to 206/207. No matter what values
I put in there, even using the same ones as reported by M503. 

And now the funny thing: After power cycling the values are loaded properly - but the extruder will
behave exactly like in the beginning - slowing down at 195, reaching 196/197 and then overshoot
to 206/207.

I enabled PID_DEBUG and added some output with my configs as a ZIP.

I tried PID_FUNCTIONAL_RANGES between 10-15, when using 15 the initial slowdown + overshoot
does happen a bit earlier, at ~192/193.

I did not see any other recent posts about this, so I'm still considering the possibility of me plainly
missing something - but I don't understand what it might be, I've been using previous releases and
can't think of a config setting I'm missing here.

Oh, by the way: While M303 Debug Output is reporting E0 and the configs are for E0 the printer
will say "E1 heating" on the display. I've noticed this a while ago but didn't really spent much time
thinking about it... but shouldn't this be "E0 heating"?

### Bug Timeline

First noticed in 2.1.1 release

### Expected behavior

I'd expect PID behavior to behave like before when setting the current values again, right
now it does look like it's using fixed values I'm unable to find... ^^

### Actual behavior

A - Power Cycle
B - Check settings using M503:
```
; Hotend PID:
M301 P42.99 I5.84 D79.10
```
C - M104 S200, slows down as described, then overshooting
E - Using M301 to set exact same values as reported by M503:
```
SENT: M301 P42.99 I5.84 D79.10
READ: ok
SENT: M500
READ: echo:Settings Stored (699 bytes; crc 49431)
Settings Stored (699 bytes; crc 49431)
READ: ok
```
F - No matter what values, it will always overshoot to 206/207 from now on until a power cycle. I also tried
completely off values like 25/2.5/125. They're saved/loaded correctly according to the menus/M503 output
but it doesn't change anything...

### Version of Marlin Firmware

bugfix-2.1.x, downloaded and configs set up today

### Printer model

DIY i3

### Electronics

SKR2 Rev B

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/9735281/configs.zip)

parallyze, 2022-12-14:
Hi all,

sorry to bring this up again - but I still suspect something wrong with PID settings in bugfix2.1.x.

I did download/configure the version from yesterday (including up to commit 314e52b) and tested
again with the same results:

1. Run PID autotune
2. Set PID values (M301) from autotune (P36.47, I4.65, D71.49 in this case),
3. Save/load with M500/M501
4. Heat hotend to 210°C using M109

It will slightly overshoot by 2-3°C, nothing I'm too woried about. But:

No matter what values I put into M301/M500/M501 from now on it
will **always** behave the same when heating up - just overshooting by 2-3°C,
even when using completely different values like P15, I4.0, D70.0. It does feel
like the values are changing absolutely nothing from here on...

...and once resetting the printer and verifiying with M503 the values are
reported correctly it will behave like written in the initial post - slow down
heating 5°C below the target and then overshoot by 4-5°C again.

Oh, and somehow I'm still irritated by "E1 heating..." (M109) on the display
when there's only E0.... at least that's what the axis configs are for. The socket
for E1 on my SKR2 Rev. B is used for Z2/G34.

Current configs attached. I can test/provide other infos if required, but right
now I'm not seeing any differences to what I initially reported here. Or what I'm
missing. But looking at more recent posts about PID Scaling and such I wonder
if this might be somehow related (not using Scaling or Autotemperature myself).

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/10230102/configs.zip)

thisiskeithb, 2022-12-14:
> Oh, and somehow I'm still irritated by "E1 heating..." (M109) on the display
when there's only E0....

Marlin uses E0 in the config and displays E1 on the LCD for readability. E1 would be E2, etc. etc.

As for your PID issues... It's not a fix, but give `MPCTEMP` a try for basically zero overshoot:

[image]

parallyze, 2022-12-14:
> Marlin uses E0 in the config and displays E1 on the LCD for readability. E1 would be E2, etc. etc.

Okay. You know those things you've seen a million times and once you start questioning if it has
been this way in the past you suddenly absolutely don't know for sure? :D

> As for your PID issues... It's not a fix, but give `MPCTEMP` a try for basically zero overshoot:

Didn't want to test this because it's still labelled "EXPERIMENTAL" (but this might change for 2.1.2
if I don't mess it with something else right now?) , but you're right and it does work without
overshoot. It does hover at 209.0 - 209.5 for quite a while but thinking about how far away the
thermistor is placed from the heater inside the hotend this looks absolutely great. After 60-90 
seconds it now stays very consistent at 209.98-210.14°C.

Let's see if somebody else can see the same ghosts I've seen on PID tuning. I've another printer
left running 2.0.8, so I'll check this on that one in the near future (and MPCTEMP).

Thanks!

parallyze, 2022-12-14:
> Several features are marked as such / have been for years and work fine. Unless you have a PTC hotend like an E3D Revo or some insane wattage heater cartridge, `MPCTEMP` works really well.

Actually I'm using a 50W cartridge, which I think is way more than this printer needs - but I had this one at hand when building the printer, so I used it (and limited it to BANG_MAX 204). 

Not really afraid of "experimental", but when it comes to handling heaters/watchdog stuff I prefer staying away from them unless there's a good reason. Opposite would be something like IS, can't wait to test it on that machine but already know it might get very interesting (TMC2208s + 24V)... :)

VanessaE, 2022-12-23:
@thisiskeithb while trying to sort out an issue with PID control on my printer, I ran across this issue.  @parallyze's problem is somewhat different from mine, but since you suggested to him/her to try MPC mode as a workaround, I decided to try it also.  To say I'm impressed would be an understatement.  
 No more hotend PID for me.  :smiley:

Grenite, 2023-02-25:
@parallyze Did you enable **#define PID_DEBUG** and configuration.h and checked the PID outputs on the serial console? I noticed after 2.9.5 the pid temps would drastically overshoot by 10C before settling. And when looking at the pid debug showed that the I value was shot up all the way to the max value of 255 when it shouldn't be - leading to heater power maximizing out when it shouldn't be. After running PID autotune I manually tuned the I term to within the 0.01 range and then I got much more reasonable heater behavior without overshoot.
I attached a file showing the PID debug output in prior to my workaround with PID_FUNCTIONAL_RANGE set to 30 to show the I term going max up until the set temp of 210C is reached.
[pid_log_subset_210_autotune_default.txt](https://github.com/MarlinFirmware/Marlin/files/10829851/pid_log_subset_210_autotune_default.txt)

parallyze, 2023-02-25:
@Grenite 
> Did you enable **#define PID_DEBUG** and configuration.h and checked the PID outputs on the serial console?

Yes and it's included in the configs.zip attached to the first post

Grenite, 2023-02-25:
@parallyze The I term in your logs show it maxed out up until the set temperature is reached. Try turning the I constant down by like 1/5 of what you had before.

parallyze, 2023-02-25:
@Grenite 

> @parallyze The I term in your logs show it maxed out up until the set temperature is reached. Try turning the I constant down by like 1/5 of what you had before.

Did you actually read the previous posts in this thread?

1 - Changing the values would always produce the same behavior, no matter WHAT values entered
2 - Power-cycling the printer would reset the behavior - while the newly set values are loaded/displayed
3 - I'm using MPCTEMP now

So if your PID settings do actually change the behavior after each change, it does not sound like the problem I
had. If your printer works correctly after setting PID _and_ turning power off/on, it also does not sound like this problem.

> I noticed after 2.9.5

I've not checked this with 2.1.2 or later running on the machines (MPCTEMP), so this might very well
have changed. Haven't been around here for a few weeks, so didn't check recent bugfix-212 builds yet,
but 2.9.5 sounds weird...?

Grenite, 2023-02-25:
@parallyze 
>1. Changing the values would always produce the same behavior, no matter WHAT values entered

Yeah I know what I suggested sounds absolutely insane and counterintuitive. I was also banging my head on the table wondering why it seemed like my PID adjustments were absolutely doing nothing until I borderline got rid of the "I" from the PID by setting it to almost 0. 

>2. - Power-cycling the printer would reset the behavior - while the newly set values are loaded/displayed

This might be from the issue with the I term - it's maxing out when it shouldn't be. I looked at my PID logs and compared with yours and it had the same behavior.

>3 - I'm using MPCTEMP now

Fashionably late to the MPCTEMP party, it must be working well for you yes? I was on 2.0.9.2 before and PID heating was rock solid for me until I upgraded to 2.1.x.

synthetic65535, 2023-05-06:
To prevent overshooting try:
1. Lowering heating power during PID
#define PID_MAX 128
2. Increase pid smoothing factor
#define PID_K1 0.5

parallyze, 2023-05-06:
> To prevent overshooting try:
> 
> 1. Lowering heating power during active PID regulation
>    #define PID_MAX 128
> 2. Increase pid smoothing (by lowering K1 factor)
>    #define PID_K1 0.5

...which doesn't do anything about the bug reported in the first post. Also PID_MAX
is set to what it is for a reason.

When Grenite started adding his problem to this thread I downloaded the current
2.1.x bugfix. Didn't have much time at hand then, so config/testing took a moment.

The problem I stated in the initial post isn't happening any longer on current
builds, my assumption would be the commits about PID in february solving this. 

Verified on a second machine which also exhibited the problem reported initially.

PID values are now stored properly and don't behave different when power cycling
the machine. Which is what this thread was about, not tuning PID in general....

closed 2023-05-06 by parallyze