MarlinFirmware/Marlin #24845: [FR] Adjustable spindle speed(Spindle Pwr) while printing

mAd-DaWg, 2022-10-08:
### Is your feature request related to a problem? Please describe.

While "printing" with a cnc machine, its often needed to be able to adjust the spindle speed on the fly. Sending a "M03 S18000" command either via the menu, or via terminal does not apply the speed change immediately, but only minutes later, potentially after issues have already occurred due to incorrect spindle speeds. Sometimes the speed change only applies when stopping a "print" via the marlin menu, causing the spindle to suddenly change speeds, which is potentially dangerous.

### Are you looking for hardware support?

No

### Describe the feature you want

Marlin should process M03 commands with priority while printing (perhaps via Emergency parsing?)

mAd-DaWg, 2022-10-08:
Example cnc configuration i am using (with btt skr2 board)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9739623/Configuration.zip)

DerAndere1, 2022-10-13:
M3 is working as documented. G-codes, including M3, are supposed to be processed sequentially. What you actually want is probably speed override and feed override. LinuxCNC has M-codes to enable feed/speed override: M48 to M53. The feed/speed multiplier is set on the fly using controls (buttons/sliders), not via M-codes.

mAd-DaWg, 2022-10-16:
@DerAndere1 you are right, that is exactly what i want. I also want to be able to apply it via the marlin menu, aka when im standing in front of the machine.