MarlinFirmware/Marlin #24846: Marlin 2.1.x bugfix issue with pins debugging on a 32 bit board

SamppaD78, 2022-10-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi. I was updating firmware from 2.9.3 to 2.1.x bugfix and came across a issue. When pins debugging is enabled i cant recompile firmware. Issue is a #define num_analog_last line, that is reporting error. Printer control board is a SKR mini E3 V2.0 with STM32F103RC controller. Only work around that i found is disabling pins debugging,after that firmware recompiled without issues.
Is there a better solution for this problem?

### Bug Timeline

started after updating to 2..1.x bugfix

### Expected behavior

Expected no issues when pins debugging is enabled

### Actual behavior

Cant compile 2.1.x bugfix firmware

### Steps to Reproduce

1. enable pins debugging  on a marlin 2.1.x bugfix and compile the firmware.

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Ender 3 pro

### Electronics

SKR mini E3 V2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/9739789/Configuration.files.zip)

cbagwell, 2022-10-08:
It looks like buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h needs same update as #24748 .  Adding this should get it compiling:

```
#define NUM_ANALOG_FIRST PA0
```

SamppaD78, 2022-10-08:
Hi. Where do i add this line, im looking at pins_arduino.h file and line 64?

cbagwell, 2022-10-08:
Add like this diff:

```
diff --git a/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h b/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h
index 8f17d052bb..54e4babe60 100644
--- a/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h
+++ b/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h
@@ -88,6 +88,7 @@ extern "C" {
 #define NUM_DIGITAL_PINS        51
 // This must be a literal with a value less than or equal to to MAX_ANALOG_INPUTS
 #define NUM_ANALOG_INPUTS       16
+#define NUM_ANALOG_FIRST        PA0
 
 // On-board LED pin number
 #ifndef LED_BUILTIN
```

SamppaD78, 2022-10-08:
Hi.I found the location thanks for help, but by adding #define NUM_ANALOG_FIRST  PA0 i get a lot of new errors regarding this added analog PA0 input. So it looks like work around doesn't work with this stm controller. Well at least i can disable debugging and get Marlin to compile.

cbagwell, 2022-10-08:
Make sure your using latest bugfix-2.1.x that contains pull request #24748 or you will get compile failures.  Are you seeing the issue with latest?

SamppaD78, 2022-10-08:
I didn't test the 2.1.1 latest version. I will test it tomorrow and report back. I was using the 2.1.x bugfix as i expected it to have all the latest fixes.

SamppaD78, 2022-10-09:
Hi. I have tested latest Marlin 2.1.1 and no problems with enabling pins debugging. Marlin compiles without errors.

thisiskeithb, 2022-10-09:
[`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x) is "bleeding edge" and should be used for all testing of potential bugs since it serves as the main branch for PRs/bug fixes & is a feeder to the `2.1.x` branch.

[`2.1.x`](https://github.com/MarlinFirmware/Marlin/tree/2.1.x) is a rolling branch of updates from `bugfix-2.1.x` above and usually lags behind it up until a release is tagged.

[`2.1.1`](https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.1) is a static `2.1.x` release/snapshot at a particular point in time and should not be used to test for bugs.

So...

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

SamppaD78, 2022-10-10:
Hi. I downloaded latest bugfix 2.1.x from provided email link and i can confirm that issue persist
when pins_debugging is enabled, all other settings are as is,only added configuration files for ender 3 pro SKR mini E3 2.0.

With 2.1.1.marlin static no issues, Marlin compiled without problems.

hcet14, 2022-10-23:
Hi,
I also have an issue with "NUM_ANALOG_LAST"

- In file included from Marlin/src/gcode/config/../../pins/pinsDebug.h:171,
                 from Marlin/src/gcode/config/M43.cpp:29:
Marlin/src/gcode/config/../../pins/../HAL/STM32/pinsDebug.h:114: warning: "NUM_ANALOG_LAST" redefined
  114 | #define NUM_ANALOG_LAST ((NUM_ANALOG_FIRST) + (NUM_ANALOG_INPUTS) - 1)
      | 
In file included from /home/hcet/.platformio/packages/framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6/cores/arduino/Arduino.h:56,
                 from Marlin/src/gcode/config/../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin/src/gcode/config/../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin/src/gcode/config/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/gcode/config/../../inc/MarlinConfig.h:31,
                 from Marlin/src/gcode/config/M43.cpp:23:
/home/hcet/.platformio/packages/framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6/cores/arduino/pins_arduino.h:65: note: this is the location of the previous definition
   65 |   #define NUM_ANALOG_LAST           (NUM_DIGITAL_PINS + NUM_ANALOG_INPUTS)

Last time I compiled marlin was Marlin-bugfix-2.1.x_220816_13_34, output was firmware.bin 179,5 KiB.

Today I compiled the actual Marlin-bugfix-2.1.x with above warning and a firmware.bin **114,9 KiB** which wouldn't boot.

I have no clue!

Maybe try '//#define PINS_DEBUGGING'?

Does anyone have an idea?

[Marlin-bugfix-2.1.x_220816_13_34.zip](https://github.com/MarlinFirmware/Marlin/files/9847254/Marlin-bugfix-2.1.x_220816_13_34.zip)
[Marlin-bugfix-2.1.x_221023_15_47.zip](https://github.com/MarlinFirmware/Marlin/files/9847255/Marlin-bugfix-2.1.x_221023_15_47.zip)
Kind regards, hcet14

SamppaD78, 2022-10-23:
Hi. Turn off pins debugging and compile firmware again.

Hanki BlueMail for Android<https://bluemail.me>

In file included from ***@***.***/cores/arduino/Arduino.h:56,

hcet14, 2022-10-23:
> Hi. Turn off pins debugging and compile firmware again.

... with //#define PINS_DEBUGGING

No change, except firmware.bin **110,2 KiB**, but still not booting.

hcet14, 2022-10-24:
@cbagwell 
I agree. Sorry I'm a little short of time right now. I'll try to get back at the weekend. 
Thank you.
Kind regards, hcet14

closed 2023-01-03 by github-actions[bot]