MarlinFirmware/Marlin #24848: CR20 Pro/SKR v1.3 LCD goes blank after flashing Marlin bugfix-2.1.x

vbezos, 2022-10-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After flashing my CR20-Pro with Marlin bugfix-2.1.x, LCD(MKS MINI12864) just went blank.
Tried to see if it is a contrast thing, but didn't have any luck.

After a while I checked the pins_BTT_SKR_V1_3.h and saw tha pin configuration was different from bugfix-2.0.x's pin file.
Then went on to replace pin file with the old one (bugfix-2.0.x), everything works fine for the display.

### Bug Timeline

A week ago

### Expected behavior

Expected to flash bugfix-2.1.x and display working properly.

### Actual behavior

Display went blank after flashing new firmware

### Steps to Reproduce

1) CR20-Pro with SKR v1.3
2)Flash bugfix-2.1.x firmware
3)Observe blank LCD screen

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

CR20-Pro

### Electronics

SKR v1.3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-10-09:
Huh?  There is no difference in pins_BTT_SKR_V1_3.h between bugfix 2.1.x and bugfix 2.0.x 

https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.0.x/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h
https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.1.x/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h

vbezos, 2022-10-09:
True story just compared the pin file i had locally from bugfix-2.0.x and that pin file is that differs, and the one working.
Problem is bugfix-2.1.x pin file is creating the blank lcd problem

[image]

Left side the pin file i had with bugfix-2.0.x

ellensp, 2022-10-09:
The pin names are the same, the plug pin numbering was rotated 180 degrees
```
 *  (BEEPER) 1.30 |10  9 | 0.28 (BTN_ENC)        (MISO) 0.17 |10  9 | 0.15 (SCK)
 *  (LCD_EN) 1.18 | 8  7 | 1.19  (LCD_RS)     (BTN_EN1) 3.26 | 8  7 | 0.16 (SD_SS)
 *  (LCD_D4) 1.20   6  5 | 1.21  (LCD_D5)     (BTN_EN2) 3.25   6  5 | 0.18 (MOSI)
 *  (LCD_D6) 1.22 | 4  3 | 1.23  (LCD_D7)   (SD_DETECT) 1.31 | 4  3 | RESET
 *            GND | 2  1 | 5V                            GND | 2  1 | --
```
became 
```
 *  (BEEPER) 1.30 | 1  2 | 0.28 (BTN_ENC)        (MISO) 0.17 | 1  2 | 0.15 (SCK)
 *  (LCD_EN) 1.18 | 3  4 | 1.19  (LCD_RS)     (BTN_EN1) 3.26 | 3  4 | 0.16 (SD_SS)
 *  (LCD_D4) 1.20   5  6 | 1.21  (LCD_D5)     (BTN_EN2) 3.25   5  6 | 0.18 (MOSI)
 *  (LCD_D6) 1.22 | 7  8 | 1.23  (LCD_D7)   (SD_DETECT) 1.31 | 7  8 | RESET
 *            GND | 9 10 | 5V                            GND | 9 10 | --
```

So
EXP1_05_PIN -> EXP1_06_PIN
EXP1_04_PIN -> EXP1_07_PIN
EXP2_09_PIN -> EXP2_02_PIN
EXP2_05_PIN -> EXP2_06_PIN

There are all fine

But the FORCE_SOFT_SPI  this would be the issue

ellensp, 2022-10-09:
Found it. This was lost with this pr https://github.com/MarlinFirmware/Marlin/pull/16349
They added a #elif ENABLED(ENDER2_STOCKDISPLAY) block and "stole" your #define FORCE_SOFT_SPI

ellensp, 2022-10-09:
I have created a PR to fix this, so closing this issue.

closed 2022-10-09 by ellensp

vbezos, 2022-10-09:
Thanks that was swift!