MarlinFirmware/Marlin #24857: [BUG] MMU2S with Octopus Pro board not responding

yannick-alch, 2022-10-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

I want to use the Prusa MMU2S with the BTT Octopus Pro board by connecting it on the TFT port.
I am using Marlin 2.1
When I start my printer, after few seconds I have "MMU not responding - DISABLED" message...
In the Configuration_Adv.h file I define #MMU2_SERIAL_PORT 1 to use on the TFT port.

I know the communication between Octopus Pro and the Prusa MMU working because I can send a soft reset to the MMU and it's working fine, but that's all.

### Expected behavior

I want the MMU responding with the Octopus Pro board

### Actual behavior

Octopus Pro board did not receive the "start" message from the Prusa MMU and the I have the message "MMU not responding - DISABLED" message"

### Version of Marlin Firmware

Marlin 2.1

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

You will find my config files below.

Thanks for your reply.

[image]

[config.zip](https://github.com/MarlinFirmware/Marlin/files/9747410/config.zip)

GMagician, 2022-10-10:
You may try to enable MMU2_DEBUG in mmu2.cpp to see if info may help

yannick-alch, 2022-10-11:
> You may try to enable MMU2_DEBUG in mmu2.cpp to see if info may help

It's already enable in Conguration_Adv.cpp

ellensp, 2022-10-11:
then how about a log ?

yannick-alch, 2022-10-11:
> then how about a log ?

Which log do you want ? This is the only message I have...

ellensp, 2022-10-11:
That is confusing then

You say "I know the communication between Octopus Pro and the Prusa MMU working because I can send a soft reset to the MMU and it's working fine"  but the error you list says its not communicating at all.

It should loo like https://revilor.github.io/MMU2-Marlin/debugging.html

yannick-alch, 2022-10-11:
[image]
My bad, I thought I activate MMU2_DEBUG...

This is what I have when I reset the board

ellensp, 2022-10-11:
as I  said, not talking at all...  

So how have you wired this?  details, pin for pin...

yannick-alch, 2022-10-11:
The wiring is like (from MMU to Octopus) :
```
Red to 5v
White (RX MMU) to pin PA9 (TX1 octopus)
Green (TX MMU) to pin PA10 (RX1 octopus)
Black shielded wire (GND MMU) to GND (octopus)
Black (reset MMU) to RST (reset pin on octopus)

```
This is the connector on octopus board
[image]

This is the connector on MMU
[image]

Hre is the pinmap of the octopus board
[image]

ellensp, 2022-10-11:
Looks ok. 

In your config you have 
#include "defines.h"
#include "config.h"
#include "change_tool.h"

what is in these files?

yannick-alch, 2022-10-11:
> Looks ok.
> 
> In your config you have #include "defines.h" #include "config.h" #include "change_tool.h"
> 
> what is in these files?

I have different printer, it's juste varaible for each machine, lige drivers, size, position for tool changing...

yannick-alch, 2022-10-11:
Is `#define MMU2_SERIAL_PORT 1` is correct with the TFT port ?

ellensp, 2022-10-11:
According to circuit diagrams, yes

yannick-alch, 2022-10-11:
Do you know if I can use another port ?

ellensp, 2022-10-11:
RJ11 is CAN1TX and CAN1RX  not TX1 RX1

yannick-alch, 2022-10-11:
Ok thanks.
I know the MMU wprking fine because I use it with GTR V1.0 board.
I don't know why it's not working with the Octopus Pro board

EvilGremlin, 2022-10-11:
maybe MMU rx/tx labels are wrong? swap wires?

flynhigh20, 2022-11-07:
Having same issues and litteraly have on tft plug aswell but will not respond, worked fine in creality board, you get fixed...

flynhigh20, 2022-12-12:
> Having same issues and litteraly have on tft plug aswell but will not respond, worked fine in creality board, you get fixed...

Mine is actually on uart2 the rpi connection using serail port 3 was working then stop maybe loose connection but ive canged nothing since

closed 2023-02-21 by github-actions[bot]