MarlinFirmware/Marlin #24860: [BUG] Dual X axis not working

Johnny-Martin, 2022-10-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

At first, I set my priner as a Dual-Y-Axis printer, it works but the printed model is mirrored. The mother board is Fysetc Spider 2.3, and here is my setting:
`#define X_DRIVER_TYPE  TMC2209_STANDALONE
#define Y_DRIVER_TYPE  TMC2209_STANDALONE
#define Z_DRIVER_TYPE  TMC2209_STANDALONE
//#define X2_DRIVER_TYPE TMC2209_STANDALONE
#define Y2_DRIVER_TYPE TMC2209_STANDALONE
#define E0_DRIVER_TYPE TMC2209_STANDALONE`
[image]
[image]

It works perfect except the mirror problem, and the y1 and y2 steper are all running while printing. To fix the mirror problem, I exchanged the x and y axis:
`#define X_DRIVER_TYPE  TMC2209_STANDALONE
#define Y_DRIVER_TYPE  TMC2209_STANDALONE
#define Z_DRIVER_TYPE  TMC2209_STANDALONE
#define X2_DRIVER_TYPE TMC2209_STANDALONE
//#define Y2_DRIVER_TYPE TMC2209_STANDALONE
#define E0_DRIVER_TYPE TMC2209_STANDALONE`
[image]
[image]

Now I have a new problem: the x2 steper is not running! I have tried every driver slot for x2(from slot E1 to E4), it never runs.

### Expected behavior

The x2 steper running when printing

### Actual behavior

The x2 steper not running

### Steps to Reproduce

1. enable x2 axis in Configuration.h
2. connect x2 steper to slot E2(I have tried from E1 to E4)
3. move X axis with pronterface
4. the x2 steper not working

### Version of Marlin Firmware

2.1.1

### Printer model

MakerBot

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-10-11:
#define X2_STEP_PIN                         PD12
#define X2_DIR_PIN                          PC4
#define X2_ENABLE_PIN                       PE8
#define X2_CS_PIN                           PA15

Ie X2 is M7

ellensp, 2022-10-11:
The model being mirrored is one of you axis is backwards
There are only two valid locations for X0,Y0   Front left of bed or back right of bed, all other combinations result in mirrored parts

This is a basic setup issue

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-10-11 by ellensp