MarlinFirmware/Marlin #24862: [FR] Seek max/min endstops for stroke validation or calibration

Varghesemela, 2022-10-11:
### Describe the feature you want

Hi team,
I am working on a seperate project where i use additional optical endstops in the X and Y axis to detect the hard positions of the gantry structure. I make use of endstops because it is possible for the motor to skip steps, maybe due to over-acceleration or overload. But if the machine is homed only once in its whole on-time, these offsets due to skipping steps can get accumulated overtime and since there is no closed loop correction, it will remain in an offset state based on the software position maintained by Marlin.
 Homing <--------------scan and stop at endstopB------------->Max bed limit (hard endstop)
endstopA ----------------------endstopB--------------------------endstopC
So what i propose is to have a special routine to reach an endstop by simply scanning the endstop along the path. This way we can refresh (or update) the absolute position of the machine in intermediate routines. Its a feature which might be helpful more in applications that use Marlin in cartesian based systems other than just 3D printers.
Another use case of this approach is to retune or calibrate the maximum available stroke of a machine. If a machine is homed and one axis, it can measure the distance travelled in that axis my reaching the max axis endstop. This way, we can calibrate the max bedsize using the machine itself.
  Homing <-----------------Distance=100.50mm------------->Max bed limit (hard endstop)
endstopA -------------------------------------------------------endstopB
 I have already written the code for this application in my project and have been using these since the last 4 months. They have worked perfectly so far and i use them to seek endstops along the way when designing new systems or replicas.

Can we add this as a feature in marlin, as i believe this can be helpful.

ellensp, 2022-10-11:
You can re home just X or X and Y at any time using gcode.

This is a slicer issue

"maximum available stroke of a machine"

This presumes the steps/mm is correct and the has been no belt stretching or other issues...

Varghesemela, 2022-10-12:
> You can re home just X or X and Y at any time using gcode.
> 
> This is a slicer issue
> 
> "maximum available stroke of a machine"
> 
> This presumes the steps/mm is correct and the has been no belt stretching or other issues...

Yes, but that presumption can be avoided i feel by introducing the proposed solution, also the other feature to get the available stroke from home position to any other position using endstops can be helpful in determining actual positions to a precision in decimals in mm.

EvilGremlin, 2022-10-12:
This is basically changing one set of assumptions for another, much wonkier one. With what you propose, i forsee layer shifts all around... It's much better to detect skipped steps with TMCs stallguard and rehome only then, like Prusa do (which is quite wonky solution too). There is only one proper way: encoders in feedback loop. Any other attempts have backfire potential. Same principle as with vaccines, inefficient one is worse that no vaccine at all.

Varghesemela, 2022-10-12:
I understand your point @EvilGremlin . I use encoders too, but since those are incremental encoders, there is always a chance of errors. Hence i use these hard endstops as zero-referencing positions so that the machine has more points to re-calibrate. This is just an additional step, a redundancy.
That is fine, i feel this feature isn't helping much in the normal working of a 3D printer, and serves only a complex system that uses marlin for cartesian mechanisms that can be used elsewhere.

closed 2022-10-12 by Varghesemela

EvilGremlin, 2022-10-13:
inremental are fine and easy to work with, you just have to eleminate rounding error possibility (don't use floats in feedback loop)