MarlinFirmware/Marlin #24864: [BUG] src/core/types.h (assignment of read-only location 'v')

murrayhenwood, 2022-10-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In src/core/types.h source file line 598, 599, 752 the code tries to assign to operation parameter. 

the lines in question 
598
FI XYZval<T>& operator*=(const float &v)             { NUM_AXIS_CODE(x *= v,    y *= v,    z *= v,    i *= v,    j *= v,    k *= v,    u *= v,    v *= v,    w *= v);    return *this; }
599  
FI XYZval<T>& operator*=(const int &v)               { NUM_AXIS_CODE(x *= v,    y *= v,    z *= v,    i *= v,    j *= v,    k *= v,    u *= v,    v *= v,    w *= v);    return *this; }
752
FI XYZEval<T>& operator*=(const T &v)                   { LOGICAL_AXIS_CODE(e *= v,    x *= v,    y *= v,    z *= v,    i *= v,    j *= v,    k *= v,    u *= v,    v *= v,    w *= v);    return *this; }

### Bug Timeline

not sure

### Expected behavior

Code to compile

### Steps to Reproduce

1. Try to compile with 8 axis 

### Version of Marlin Firmware

2.1.X

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No configuration needed after reading the code it should be clear about the problem

ellensp, 2022-10-12:
How about those configs so we don't have to waste time recreating this?

murrayhenwood, 2022-10-12:
Sorry, There are no boards have the pin definitions needed to show the problem and therefor there is no simple configuration and changes that could be zipped up. 
This is obviously a niche problem, feel free to close the issue.

EvilGremlin, 2022-10-12:
Then link to your fork or something. If fixing this could save us feature headaches it's worth looking at. Making Marlin more flexible and scalabe (both up and down) is ever-present goal.

murrayhenwood, 2022-10-12:
Roughly I think these are the minimum changes needed, using Platform IO for the build
https://github.com/murrayhenwood/Marlin

ellensp, 2022-10-12:
PR https://github.com/MarlinFirmware/Marlin/pull/24866 should fix it

closed 2022-10-12 by ellensp