MarlinFirmware/Marlin #24868: Using 3 Z-motors, G34 Fails and nozzle crashes to bed

mattiras, 2022-10-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using 3 Z-motors, G34 moves to first measuring point and starts to run towards the bed, but gives G34 failed error immediately and runs nozzle to the table and keeps trying to run the z-motors indefinitely.

G28 and G29 works as expected. Going back to Marlin 2.0.8.2 works. With 2 Z-motors works.

### Bug Timeline

Since 2.0.9.

### Expected behavior

a) G34 command to finish
b) When failing, all movement to stop.

### Actual behavior

Starts but in first measuring point outputs error and runs nozzle to the bed and keeps trying to run the z-motors indefinitely.
Behavior is strange compared to successfull G34.

### Steps to Reproduce

1. uncomment Z_STEPPER_AUTO_ALIGN
2. use 3 Z-motors
3. Use command G34

### Version of Marlin Firmware

2.1.1

### Printer model

DIY coreXY

### Electronics

Ramps 1.4

### Add-ons

3xZ drivers A4988. Z2 on E1 slot, Z3 driver modded on top of Z1

### Bed Leveling

ABL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[G34 failing.zip](https://github.com/MarlinFirmware/Marlin/files/9778965/G34.failing.zip)

thinkyhead, 2022-10-14:
Does it help to set `Z_PROBE_LOW_POINT` to something like -10 ?

mattiras, 2022-10-14:
> Does it help to set `Z_PROBE_LOW_POINT` to something like -10 ?
I Tried earlier and that has no affect.

closed 2024-06-21 by github-actions[bot]