MarlinFirmware/Marlin #24869: [BUG] Marlin 2.1.1/2.1.x Trigorilla_Pro reboots when screen is touched

Frontier314, 2022-10-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was updating from Marlin 2.0.9.3 on my Trigorilla_Pro board (the printer is Anycubic Predator) going all the way up to 2.1.x bugfix (2.0.9.3 -> 2.0.9.4 -> 2.0.9.5 -> 2.1.1 -> 2.1.x). Using the provided example configuration files for Anycubic Predator (config/examples/delta/Anycubic/Predator).

Starting from 2.0.9.4, neither the standard or maple environments produce a working firmware; the source compiles without errors but the board does not boot at all. If the following change is done to the stm32f1-maple.ini on 2.0.9.4, the board works as expected:

```
[env:trigorilla_pro_maple]
extends              = STM32F1_maple
board                = genericSTM32F103ZE
board_build.address  = 0x08000000
build_flags          = ${STM32F1_maple.build_flags} -DSS_TIMER=4 -DSTM32_XL_DENSITY

```
The above change is all that is needed to make maple environment produce a working firmware for 2.0.9.4, 2.0.9.5, 2.1.1 and 2.1.x.

The problem is that when I touch the touchscreen in order to navigate the menu system, the board reboots. This behaviour started from 2.0.9.5 and goes up to 2.1.x bugfix. On 2.0.9.4 when the user touches the screen, the board makes a clicking sound with the onboard beeper and it is possible to navigate the menus without any issues (and no crashes). Board is accessible from serial.

### Bug Timeline

Started on 2.0.9.5

### Expected behavior

I expect to be able to touch the touchscreen and navigate the menus without the board rebooting.

### Actual behavior

As soon as I touch the touchscreen, the motherboard reboots and it is impossible to enter the menu system.

### Steps to Reproduce

1. Download any Marlin version from 2.0.9.5 onwards and the configuration files for the version you are going to compile from config/examples/delta/Anycubic/Predator.
2. Change the trigorilla_pro_maple definition (ini/stm32f1-maple.ini) to the following:
```
[env:trigorilla_pro_maple]
extends              = STM32F1_maple
board                = genericSTM32F103ZE
board_build.address  = 0x08000000
build_flags          = ${STM32F1_maple.build_flags} -DSS_TIMER=4 -DSTM32_XL_DENSITY

```
3. Compile Marlin using the trigorilla_pro_maple environment.
4. Flash the firmware.bin to the Trigorilla Pro board using STM Demonstrator GUI 2.8.0.
5. Reboot the board and as soon as the Marlin status screen appears, touch any menu button on the touchscreen and the board will reboot.

### Version of Marlin Firmware

2.1.x

### Printer model

Anycubic Predator

### Electronics

Stock board and printer

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[trigorilla_pro_config.zip](https://github.com/MarlinFirmware/Marlin/files/9780284/trigorilla_pro_config.zip)

rq3, 2022-10-16:
I installed the latest bugfix today (October 16, 2022) on my Anycubic Predator with no problems, other than having to fix the configuration file issues that have always been there.

Frontier314, 2022-10-17:
> 

What configuration: standard or maple?

rq3, 2022-10-17:
> > 
> 
> What configuration: standard or maple?

Standard. Complete latest bugfix attached, with hand edited addition of the Color_UI remaining time files, plus things unique to my printer (0.9 degree steppers, RTD temp sensor, mod
[Marlin_2.1.2_101522.zip](https://github.com/MarlinFirmware/Marlin/files/9802398/Marlin_2.1.2_101522.zip)
fied progress bar, etc.), so be careful. But should at least let you know if your touchscreen is OK.

Frontier314, 2022-10-18:
> 

Thank you for providing the complete bugfix release.
I was able to compile it and install it on my (spare) board but on boot it crashed (PRINTER HALTED) due to E0 sensor (since it was not the default one). Changed the configuration files to match the "stock" examples (regarding sensors) and the board booted.

Immediately after boot it enters the touchscreen calibration wizard where the firmware advanced by itself to BOTTOM LEFT point and it is impossible to click inside the crosshairs. Calibration fails after a while (it keeps "tapping" by itself) and the firmware lands on the info status screen where nothing is clickable :(

rq3, 2022-10-18:
> Thank you for providing the complete bugfix release. I was able to compile it and install it on my (spare) board but on boot it crashed (PRINTER HALTED) due to E0 sensor (since it was not the [...]

Can you run M995 (touchscreen calibration) over USB from a terminal like Repetier Host? It sounds very much like you have a faulty touch screen, or perhaps a bad cable/connection between the touchscreen and the Trigorilla Pro.

Frontier314, 2022-10-19:
> Can you run M995 (touchscreen calibration) over USB from a terminal like Repetier Host? It sounds very much like you have a faulty touch screen, or perhaps a bad cable/connection between the touchscreen and the Trigorilla Pro.
> 
> The behavior you are seeing is "normal", in that if the screen is being touched when the board is booted, it will go into touch calibration. Definitely points to a faulty touchscreen/cable/connection.

I can initiate the touchscreen calibration routine with M995 but the result is the same: touch points do not respond.

I cannot understand how the touchscreen/Trigorilla/cable can be at fault when on EXACTLY the same hardware Marlin 2.0.9.3 runs fine (100% working touchscreen) and the Predator stock firmware too. I can use the touchscreen starting from Marlin 2.0.5.8 (the first one that supported Trigorilla Pro) up to 2.0.9.3 without ANY issues with the touchscreen. Starting from Marlin 2.0.9.4 onwards, the touchscreen is not responding.

This behavior for me has to do with firmware not supporting properly the hardware.
It could be possible that I have a very late revision of the touchscreen that is not exactly the same like the one that Marlin supports but this does not explain why Marlin 2.0.9.3 runs perfectly fine on it.

> 
> If you decide that the screen is actually bad, and have nothing left to lose, I have occasionally "saved" a touchscreen by appyling a vacuum cleaner (with no brush) directly to the face of the screen to "suck" the touch surface away from the internal structure. But that's a last gasp effort, and when it works, it doesn't work for very long (a few months at most).

I will probably stick with the stock firmware and re-install Marlin 2.0.9.3 on it later.

rq3, 2022-10-19:
I had missed the part where everything works fine on earlier firmware (2.0.9.3). That's weird.

FusioNdotEXE, 2022-11-04:
> I installed the latest bugfix today (October 16, 2022) on my Anycubic Predator with no problems, other than having to fix the configuration file issues that have always been there.

I installed Marlin 2.1.x for the very first time ever personally, and first time on to my stock Anycubic Predator trigorilla pro also with no problems. By the way @rq3 , thanks for all the comments/posts on the thingiverse group page, it's what made my first attempt work out.

Frontier314, 2022-11-04:
> I installed Marlin 2.1.x for the very first time ever personally, and first time on to my stock Anycubic Predator trigorilla pro also with no problems. By the way @rq3 , thanks for all the comments/posts on the thingiverse group page, it's what made my first attempt work out.

Which configuration, standard or maple?
I still cannot understand what is wrong with my setup with the stock config files and the touchscreen, as the same setup works fine with Marlin 2.0.9.3.

Frontier314, 2022-11-18:
> I had missed the part where everything works fine on earlier firmware (2.0.9.3). That's weird.
> 
> Edit: I just tried compiling the latest bugfix under maple. It compiles, and it loads, but the screen is blank for all ui flavors (classic, color, lvg). Color_ui works fine under the standard environment.

After trying lots of things and options, I've managed to track down the issue.
The reason the maple build was rebooting the board each time the screen is touched, was related to the SPEAKER option.
If the option is disabled, the display works fine (on maple environment) without rebooting the board.
The touchscreen is still behaving erratically with the standard build environment but works fine with maple, like it did with 2.0.9.3; the only difference is now that I do not have audible feedback on screen touches.
[AnycubicPredator_Marlin2.1.x-config.zip](https://github.com/MarlinFirmware/Marlin/files/10043293/AnycubicPredator_Marlin2.1.x-config.zip)

closed 2022-11-19 by Frontier314