MarlinFirmware/Marlin #24874: [BUG] Wrong servo pins enumerating in pins_ZRIB_V53.h

therocketelectrician, 2022-10-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The servo 0 command doesn't work on my BLtouch.

### Bug Timeline

I discover the bug, because I'm installing a BLTouch on my 3D printer and upgrading marlin to the most recent version

### Expected behavior

The servos pins are put on pins A11 and A12, but writting like it is right now, the code is compile as if the servo were on pin D11 and D12.
`//
// Servos / XS3 Connector
//
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          11  // Analog Output
#endif
#ifndef SERVO1_PIN
  #define SERVO1_PIN                          12  // Analog Output
#endif`

The pin on the Bltouch doesn't move on those command:
`M280 P0 S10 ; pushes the pin down
M280 P0 S90 ; pulls the pin up
M280 P0 S120 ; Self test – keeps going until you do pin up/down or release alarm
M280 P0 S160 ; Release alarm`

### Actual behavior

I solve the problem. I just want the modification to be push on the repo.
To solve the problem, we just need to change those line:

`//
// Servos / XS3 Connector
//
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          11  // Analog Output
#endif
#ifndef SERVO1_PIN
  #define SERVO1_PIN                          12  // Analog Output
#endif`

to this:

`//
// Servos / XS3 Connector
//
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          65  // A11 Analog Output
#endif
#ifndef SERVO1_PIN
  #define SERVO1_PIN                          66  // A12 Analog Output
#endif`

### Steps to Reproduce

1-Have a RAMPS style board.
2-Select ZRIB_5.3V board
3-Connect a BL-Touch to the A11/D65 pins
4-Install firmware
5-Use the command: M280 P0 S120
6-See that the pin of the Bltouch is not moving
7-Change the pin in pins_ZRIB_V53.h from 11 to 65
8-Reupload
9-It is working!

### Version of Marlin Firmware

Bugfix-2.1.x - 15 october 2022

### Printer model

Homemade cartesian

### Electronics

ZRIB V5.3

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/9792703/config_files.zip)

ellensp, 2022-10-16:
Are you sure you not on the wrong plugs?  (I dont have access to this board to check) 

From the pins file
 ```
* XS3 Connector
 *  =================
 *  | 65 | GND | 5V |      (65)  PK3 ** Pin86 ** A11
 *  |----|-----|----|
 *  | 66 | GND | 5V |      (66)  PK4 ** Pin85 ** A12
 *  =================
 * XS3/Servos Connector
 *  =================
 *  | 11 | GND | 5V |      (11)  PB5 ** Pin24 ** PWM11
 *  |----|-----|----|
 *  | 12 | GND | 5V |      (12)  PB6 ** Pin25 ** PWM12
 *  =================
```

ellensp, 2022-10-16:
found this on the internet

https://img.tttcdn.com/product/xy/2000/2000/p/gu1/O/6/OS0516/OS0516-1-4ead-lp7K.jpg

seems like your correct

ellensp, 2022-10-16:
Pr created. Closing issue

closed 2022-10-16 by ellensp