MarlinFirmware/Marlin #24881: STM32 Hardware Requirements

darkomenz, 2022-10-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I am looking for information on what silicon requirements are required to run marlin on the STM32.

What would be the recommended CPU & Flash requirements (Architecture, Clock Speed, Ram Size, Flash Size)?
Do I have to use the hardware SPI peripheral or is there software SPI support?
Do I have to use the hardware USART peripheral or is there software USART support?
Do I have to use the hardware I2C peripheral or is there software I2C support?
Do I have to use any of the hardware PWM peripheral to drive the step pins?
Are there any benefits for using the hardware peripheral's over a software implementation?
What is the maximum number of steppers / axis that are supported?

Regards,
Brennen

### Version of Marlin Firmware

General Hardware Question

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-10-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-10-16 by ellensp

darkomenz, 2022-10-16:
@ellensp 

This is a hardware development question.
The questions being asked are valid as they are not well documented in other places on the internet.

Regards

ellensp, 2022-10-16:
Has been answered on discord.