MarlinFirmware/Marlin #24882: Marlin 2.1.2 extruder motor dont move on board MKS robin e3d

matejc88, 2022-10-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With marlin 2.0.9.2 there is no problem.
After install marlin 2.1.2 or 2.1.1 , E0 extruder motor dont move. Not when printing or sending command.
Board is MKS robin E3D v1.1 with tmc2209 with only one extrudor.

### Bug Timeline

New bug. With verison 2.0.9.2, there is no problem. Problem start with verison 2.1.1

### Expected behavior

Extrudor motor shud work.

### Actual behavior

Extrudor motor doesnt work.

### Version of Marlin Firmware

02010200

### Printer model

Anet a8 plus

### Electronics

Stock electronic

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

This is configuration, where extrudor work:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9795351/Configuration.zip)

This is configuration where extrudor dont work:

[Configuration 2.1.2.zip](https://github.com/MarlinFirmware/Marlin/files/9795356/Configuration.2.1.2.zip)

thinkyhead, 2022-10-16:
Have you gone through troubleshooting with help from the community to isolate the problem and confirm that it is not due to misconfiguration?

For help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

matejc88, 2022-10-16:
I also posted problem on facebook group. Wainting for respond.

But i have no idea what misconfiguation shud i do , that extruder dont work...

I double check all two configuration files with comprare tool with notepad++...

ellensp, 2022-10-17:
Setting a basic config for your board under bugfix 2.1.x  
BOARD_MKS_ROBIN_E3D_V1_1
#define SERIAL_PORT 1
MKS_MINI_12864_V3
NEOPIXEL_LED
NEOPIXEL_TYPE          NEO_RGB
NEOPIXEL_PIXELS              3
PINS_DEBUGGING

E works as expected 

Provided 2.1.2 config, causes bootloop with a watchdog timeout,

ellensp, 2022-10-17:
Disabling sdsupport and E comes back. 
So this gives us a starting point.

ellensp, 2022-10-17:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24380

closed 2022-10-17 by ellensp

matejc88, 2022-10-17:
I was looking on google for my problem and i didnt found this problem on github.... I apologize. Looks like i didnt used key words.

Sow, i can use SD card or e-motor. Both together still wont work. 
Tryed changing from 3 to 2 -set -DDEFAULT_SPI=2 and enable  #SDSUPPORT  but when having SD card in sreen sdslot, printer freezes.

ellensp, 2022-10-17:
This is not a final solution but it should fix it for you  (need to do more testing for other displays)

[changes.zip](https://github.com/MarlinFirmware/Marlin/files/9799514/changes.zip)

These changes are for latest bugfix 2.1.x 

It contains updated
Marlin/Configuration.h 
Marlin/Configuration_adv.h
Marlin/src/pins/stm32f1/pins_MKS_ROBIN_E3_common.h 
ini/stm32f1.ini 

If using bugfix you need to use all the updated files (config files are slightly different) 
If your using the older release just pins_MKS_ROBIN_E3_common and stm32f1.ini  should be sufficient to get running

Important for some reason you don't have neopixels enabed in your config...  this is needed for the lcd backlight
I enabled in the 2.1.x bugfix configs  

 There is still some quirkiness with the sd support  and BROWSE_MEDIA_ON_INSERT
It will time out to main status display and say Media removed, when it hasn't been removed.

matejc88, 2022-10-17:
Ok. Thank you. I will test it toonight.

Looks like i enabled this option after i send file or forgot to save file when enabled neopixels.

matejc88, 2022-10-18:
I used new Marlin/src/pins/stm32f1/pins_MKS_ROBIN_E3_common.h
and ini/stm32f1.ini in my marlin 2.1.2

Looks OK now:
SD card OK
Extruder OK
Alll OK :)