MarlinFirmware/Marlin #24883: [Change request] Please don't use "funny" ascii codes (no offence)

hcet14, 2022-10-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin is used all over the world and some languages use "funny" ascii codes (no offence).

I tried to compile Marlin-bugfix-2.1.x_221016_19_44. When I compared my old Configuration_adv.h with the new one, I stumbled over line 4062.

It said "Průša". The old one said "Prusa".

My question is, can we stay with "Prusa" (and maybe other words with "funny" ascii codes) to keep things more understandable and easy to search?

### Bug Timeline

recognized 2022/10/16

### Expected behavior

change Průša to Prusa and other words used in marlin with "funny" ascii codes

### Actual behavior

na

### Steps to Reproduce

na

### Version of Marlin Firmware

na

### Printer model

na

### Electronics

na

### Add-ons

na

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-10-17:
The guys name is Průša, he is Czech, these are the correct characters to use.

closed 2022-10-17 by ellensp

hcet14, 2022-10-23:
> The guys name is Průša, he is Czech, these are the correct characters to use.

I guess everybody, who is a bit interrested in 3D printing knows Josef Průša and maybe where he is from.

He named his book 'Basics of 3D printing with Josef Prusa' not using the 'funny' ascii codes.

No point to argue about this, way too minor. From my point of view, it's a little bit sad, that you closed it so soon. Might have been, that other users have a similar feeling?

Please don't get me wrong. I really appreciate your good work and of course the work from all the other marlin makers. Thank you all.

Kind regards, hcet14