MarlinFirmware/Marlin #24885: Anycubic Mega Pro locks constant reboots and lockup when enabling UBL. Bilinear seems to boot fine.

SidSkiba, 2022-10-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Issues with UBL on Trigorilla Mega 2650 Board. When I enable UBL and download firmware the board resets itself like 8 times then locks up. If I turn off UBL and enable Bilinear it seems to boot fine after updating firmware.

### Bug Timeline

Unknown as this is first attempt to update firmware from stock.

### Expected behavior

Expected a normal bootup after firmware update.

### Actual behavior

Multiple reboots and lockup.

### Steps to Reproduce

Enable UBL and download to board.

### Version of Marlin Firmware

2.1.1 Bugfix

### Printer model

Anycubic Mega Pro 

### Electronics

Stock with addition of BLTouch

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/9796243/Configuration.Files.zip)

Note the config set for Bilinear in upload. Comment out Bilinear and Enable UBL for testing.

tombrazier, 2022-10-18:
This kind of behaviour is consistent with running out of memory. How much SRAM does your firmware image use for the two different options?

ellensp, 2022-10-18:
@tombrazier I agree, this is the second time I seen this recently.  people adding so many options on a 4k ram system.

mega2560 needs a stack vs heap crash check.

SidSkiba, 2022-10-18:
This is what I get with compile:

With Bilinear Selected
RAM:   [========  ]  80.6% (used 6601 bytes from 8192 bytes)
Flash: [=====     ]  54.9% (used 139408 bytes from 253952 bytes)
Building .pio\build\mega2560\firmware.hex

With UBL Selected
RAM:   [==========]  97.9% (used 8024 bytes from 8192 bytes)
Flash: [======    ]  64.9% (used 164778 bytes from 253952 bytes)
Building .pio\build\mega2560\firmware.hex

SidSkiba, 2022-10-18:
Is it possible the stock Anycubic bootloader on the Trigorilla could be creating this issue?

EvilGremlin, 2022-10-18:
yep, not enought free RAM for UBL, disable something

SidSkiba, 2022-10-19:
If I turn off Linear Advance then UBL boots up. I never ran into this before with 2.0 and a Zonestar Mega 2560 board running UBL and Linear Advance (granted the display was a simple 4 line unit which may also reduce memory). Maybe 2.1 is just more memory intensive than 2.0 was.

SidSkiba, 2022-10-20:
Here is the result from 2.0.9.3 for the Zonestar board. A couple differences I notes were the display, I was using a different bed probe for that one, and I see meat packing is off (unsure why I had it on for the anycubic). I did try turning off on the Anycubic and it made little diff anyway.

RAM:   [========  ]  78.8% (used 6452 bytes from 8192 bytes)
Flash: [========= ]  89.1% (used 226278 bytes from 253952 bytes)
Building .pio\build\mega2560\firmware.hex

I have not tried changing displays for the Trigorilla yet to see if that makes a difference). Will also need to try changing the BL Touch.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9825453/Marlin.zip)

SidSkiba, 2022-10-20:
I tried changing BL touch to fixed probe and disabling meat packing. I also turned off laser feature and it made very little difference. I change the display to Zonestar and it did reduce the ram usage. 

RAM:   [========= ]  88.7% (used 7264 bytes from 8192 bytes)
Flash: [========  ]  83.2% (used 211228 bytes from 253952 bytes)
Building .pio\build\mega2560\firmware.hex

tombrazier, 2022-10-20:
Hmm, when I try building 2.0.9.3 with the config you posted I get
```
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:820:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h."
```

There is a BOARD_ZRIB_V52 in 2.0.9.3 but apparently not a BOARD_ZRIB_V53. I have built it with 2.0.9.4, though, and get a similar size.

If I port that config to 2.1.x, it's even smaller
```
RAM:   [========  ]  77.1% (used 6318 bytes from 8192 bytes)
Flash: [========= ]  87.8% (used 223002 bytes from 253952 bytes)
```

But there are stacks of changes between this config and the one in the original post.

SidSkiba, 2022-10-21:
Yeah sorry. I contributed the ZRIB V53 board and I think it was merged in 2.0.9.4. It was not in the 2.0.9.3 release. The issue is simply the Anycubic display. It takes too much memory to use UBL. I will run with Bilinear as it allows me to keep the display. Thank you for the help.

tombrazier, 2022-10-21:
There are also a lot of small extra features which, cumulatively, add up to about 10% of SRAM and then the changes to `MAX_CMD_SIZE`, `BUFSIZE`, `TX_BUFFER_SIZE` and `RX_BUFFER_SIZE` account for another 10%.

tombrazier, 2022-10-21:
I think this PR can be closed as there is no significant SRAM usage difference between 2.0 and 2.1.

closed 2022-10-21 by thisiskeithb

SidSkiba, 2022-10-21:
Agreed